convertersystems/opc-ua-samples #1: Cannot compile sample

iquestor, 2016-08-04:
I am trying to build a simple OPCUA Browser, and am considering this OPCUA Client, but I cannot get the sample code to run. I am using a VS2015 UPD 3 with a new Console App. 

I imported the NuGet package COnverterSystems.OpcUA v 5.0.0

//declaration: 
using System;
using System.Linq;
using System.Threading.Tasks;
using Opc.Ua;

Issue: these aren't recognized and I don't know the proper way to include them in a new console app:
using Workstation.ServiceModel.Ua;
using Workstation.ServiceModel.Ua.Channels;

errors:

Error	CS0103	The name 'TransportProfileUris' does not exist in the current context	ConsoleOPCUA	C:\Users\user\Documents\Visual Studio 2015\Projects\ConsoleOPCUA\ConsoleOPCUA\Program.cs	44	Active
Error	CS0103	The name 'UaTcpDiscoveryClient' does not exist in the current context	ConsoleOPCUA	C:\Users\user\Documents\Visual Studio 2015\Projects\ConsoleOPCUA\ConsoleOPCUA\Program.cs	46	Active
Error	CS0246	The type or namespace name 'UaTcpSessionChannel' could not be found (are you missing a using directive or an assembly reference?)	ConsoleOPCUA	C:\Users\user\Documents\Visual Studio 2015\Projects\ConsoleOPCUA\ConsoleOPCUA\Program.cs	57	Active
Error	CS1061	'ApplicationDescription' does not contain a definition for 'GetCertificate' and no extension method 'GetCertificate' accepting a first argument of type 'ApplicationDescription' could be found (are you missing a using directive or an assembly reference?)	ConsoleOPCUA	C:\Users\user\Documents\Visual Studio 2015\Projects\ConsoleOPCUA\ConsoleOPCUA\Program.cs	57	Active

iquestor, 2016-08-04:
Update: I did include the "Workstation.ServiceModel" Files however it didn't make a difference.

awcullen, 2016-08-04:
Please delete Nuget package "ConverterSystems.OpcUA v 5.0.0".
Install instead the package "Workstation.UaClient" which is reformulated for both UWP and WPF. 

Sorry for the confusion. And thanks for considering this project and for reporting the issue!
Andrew

iquestor, 2016-08-04:
Hi Andrew

I tried doing this, but got the error:

Severity

Code

Description

Project

File

Line

Suppression State

Error

Could not install package 'Workstation.UaClient 1.2.1'. You are trying to install this package into a project that targets '.NETFramework,Version=v4.5.2', but the package does not contain any assembly references or content files that are compatible with that framework. For more information, contact the package author.

0

iquestor, 2016-08-04:
I see the dependency is .NET 4.6; I enabled this and it appears to load.  ( I assumed VS2015 would pick most recent .NET for the project, but I guess not)

iquestor, 2016-08-04:
setting project to .NET 4.6 and deleting the ConvertSystems.OpcUA NuGET fixed the issue. thanks.

closed 2016-08-04 by iquestor