convertersystems/opc-ua-samples #5: node-opcua OPC Server certificate file location

abrasat, 2016-12-15:
I tried to start the Workstation.NodeServer, but following error is displayed:

AssertionError: fs.existsSync(this.certificateFile), "Certificate file must exist :" + this.certificateFile

In what folder does the OPC Server search the "server_selfsigned_cert_2048.pem" and "server_key_2048.pem" certificate files ?

awcullen, 2016-12-15:
\NodeServer\certificates

To generate certificates:
1.Right click project Workstation.NodeServer in Visual Studio. Select 'Open Command Prompt Here...' 
2.Execute 'node node_modules\node-opcua\bin\crypto_create_CA.js demo'

abrasat, 2016-12-15:
Thank you for the reply.
But the client certificates needed for the OPC client component are located in another location, it seems here:
\NodeServer\node_modules\node-opcua\certificates\

awcullen, 2016-12-15:
I see. You are correct!

But if you run 
node node_modules\node-opcua\bin\crypto_create_CA.js demo
then you also get them in \NodeServer\certificates

I plan to write a dedicated script to create a certificate with the correct alt subject name (from the ApplicationUri).

abrasat, 2016-12-16:
Ok, thanks.
One more question. Where does the node-opcua server store the list with users and passwords ?

awcullen, 2016-12-16:
For now, the node-server does not store a list of users and passwords. It accepts everyone.
RobotApp does store the last username:
```
Windows.Storage.ApplicationDataContainer roamingSettings = 
    Windows.Storage.ApplicationData.Current.RoamingSettings;
roamingSettings.Values[this.userNameKey] = this.userNameTextBox.Text;
```

closed 2017-01-29 by awcullen