convertersystems/opc-ua-samples #6: System.TypeLoadException: Could not load type 'Workstation.ServiceModel.Ua.Subscription[]' from assembly 'Workstation.UaClient, Version=1.5.0.30692

CarlCrol, 2016-12-22:
Where you able to make it work? I tried doing a sample application but in this line of code.

` private string discoveryUrl = @"opc.tcp://10.164.232.82:62547"; // Use ip address ( or '10.0.2.2' for accessing local host computer from emulator. hostname, localhost or 127.0.0.1 will not work on Android emu!)

        private ILoggerFactory loggerFactory;
        private ILogger logger;
        private UaTcpSessionClient session;
        protected override void OnCreate (Bundle bundle)
		{
			base.OnCreate (bundle);

			global::Xamarin.Forms.Forms.Init (this, bundle);
            // Setup a logger.
            this.loggerFactory = new LoggerFactory();
  
            this.logger = this.loggerFactory.CreateLogger<MainActivity>();

            try
            {
                // Create the session client for the app.
                this.session = new UaTcpSessionClient(
                    new ApplicationDescription()
                    {
                        ApplicationName = "Workstation.MobileHmi",
                        ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:Workstation.MobileHmi",
                        ApplicationType = ApplicationType.Client
                    },
                    new DirectoryStore(
                        Path.Combine(System.Environment.GetFolderPath(System.Environment.SpecialFolder.LocalApplicationData), "pki"),
                        loggerFactory: loggerFactory),
                    this.ProvideUserIdentity,
                    this.discoveryUrl,
                    this.loggerFactory);

                this.LoadApplication(new App(this.session));
            }
            catch (Exception ex)
            {
                this.logger.LogError("Error creating UaTcpSessionClient. {0}", ex.Message);
            }
        }
`

ex:

> {System.TypeInitializationException: The type initializer for 'System.Collections.Generic.List`1' threw an exception. ---> System.TypeLoadException: Could not load type 'Workstation.ServiceModel.Ua.Subscription[]' from assembly 'Workstation.UaClient, Version=1.5.0.30692, Culture=neutral, PublicKeyToken=null'.
  at System.Collections.Generic.List`1[T]..cctor () [0x00000] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/collections/generic/list.cs:47 
  --- End of inner exception stack trace ---
  at (wrapper managed-to-native) System.Object:__icall_wrapper_mono_generic_class_init (intptr)
  at System.Collections.ObjectModel.Collection`1[T]..ctor () [0x00006] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/collections/objectmodel/collection.cs:28 
  at System.Collections.ObjectModel.ObservableCollection`1[T]..ctor () [0x0000b] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/System/compmod/system/collections/objectmodel/observablecollection.cs:50 
  at Workstation.ServiceModel.Ua.UaTcpSessionClient..ctor (Workstation.ServiceModel.Ua.ApplicationDescription localDescription, ICertificateStore certificateStore, System.Func`2 userIdentityProvider, System.String endpointUrl, ILoggerFactory loggerFactory, Double sessionTimeout, UInt32 timeoutHint, UInt32 diagnosticsHint, UInt32 localReceiveBufferSize, UInt32 localSendBufferSize, UInt32 localMaxMessageSize, UInt32 localMaxChunkCount) [0x0002b] in <filename unknown>:0 
  at App1.Droid.MainActivity.OnCreate (Android.OS.Bundle bundle) [0x0002e] in D:\Files\Personal\Development\OpcUa\App1\App1\App1.Droid\MainActivity.cs:37 }

stacktrace:  

> at (wrapper managed-to-native) System.Object:__icall_wrapper_mono_generic_class_init (intptr)  at System.Collections.ObjectModel.Collection`1[T]..ctor () [0x00006] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/mscorlib/system/collections/objectmodel/collection.cs:28   at System.Collections.ObjectModel.ObservableCollection`1[T]..ctor () [0x0000b] in /Users/builder/data/lanes/3415/7db2aac3/source/mono/external/referencesource/System/compmod/system/collections/objectmodel/observablecollection.cs:50   at Workstation.ServiceModel.Ua.UaTcpSessionClient..ctor (Workstation.ServiceModel.Ua.ApplicationDescription localDescription, ICertificateStore certificateStore, System.Func`2 userIdentityProvider, System.String endpointUrl, ILoggerFactory loggerFactory, Double sessionTimeout, UInt32 timeoutHint, UInt32 diagnosticsHint, UInt32 localReceiveBufferSize, UInt32 localSendBufferSize, UInt32 localMaxMessageSize, UInt32 localMaxChunkCount) [0x0002b] in <filename unknown>:0   at App1.Droid.MainActivity.OnCreate (Android.OS.Bundle bundle) [0x0002e] in D:\Files\Personal\Development\OpcUa\App1\App1\App1.Droid\MainActivity.cs:37

awcullen, 2016-12-22:
Works in the emulator set for Android 7.1 - API 25 (also tested Android 6.0 - API 23). 
[image]

I'm connecting to UnifiedAutomation's CPP Demo server on a second pc. I had to set the client's EndpointUrl to remote pc's IPAddress instead of the host name because of WINS issues on the emulator.

The first attempt won't connect. Open the UA Server Administrator on the remote pc and TRUST the 'Workstation.MobileHmi' certificate.
[image]

CarlCrol, 2016-12-22:
Deploying it in Mobile phone with Andoroid API Level 23 does not work. it throws an exception still.

Have you tried debugging it in a mobile phone?

awcullen, 2016-12-22:
Would you zip up your OpcUaMobileTest solution and post it in DropBox again?

I will get my phone back when my son comes home from college ;)

CarlCrol, 2016-12-22:
okay will send you the test application that i'm using

CarlCrol, 2016-12-22:
https://www.dropbox.com/s/vcw0xld5zgvl89d/App1.zip?dl=0

awcullen, 2016-12-27:
I tested my sample "RobotApp" on a Samsung phone running Android 6.0 (Api 23). The app connects to a Unified Automation's C++ Demo Server on the local intranet. It is using security profile SHA256 sign and encrypt. 
Two items to note:
I connected the USB cable from my PC to the phone and activated USB Debugging. This involved tapping the Build Number seven times to reveal the "Developer Options" tab. See https://developer.xamarin.com/guides/android/getting_started/installation/set_up_device_for_development/. Then the Samsung became available as a debugging Target.

As always, you have to remember on the first run to tell the OPC UA Server to trust the client certificate. For the UA CPP Demo Server, this involves opening the UA Admin Dialog, finding the new certificate, and trusting it.

awcullen, 2016-12-28:
I tested your sample "App1" on the Android Emulator. I had to make these changes:
First, I had use the "OPC UA Technology Samples Welcome Dashboard"  to start the "DA Server".

Second, I had to change the url in "MainActivity.cs" to 10.0.2.2, because the server was running on the same pc as the emulator.
private string discoveryUrl = @"opc.tcp://10.0.2.2:62547/Quickstarts/DataAccessServer";

Third, I ran the app on the simulator. Initially the app did not connect to the server because the app certificate was not trusted. I found the server put the app certificate in the directory:
C:\ProgramData\OPC Foundation\CertificateStores\RejectedCertificates\certs

Fourth, I moved the .der file from the above  directory to the following:
C:\ProgramData\OPC Foundation\CertificateStores\UA Applications\certs

Finally, the server accepted the app certificate and allowed the connection:
[image]

CarlCrol, 2017-01-02:
On your test in the Samsung Mobile App, what IP Address you used for the discoverUrl?

CarlCrol, 2017-01-02:
when debugging in Samsung S6 API 23, i can't see the certificate in the refected certficate store. I'm using the DataAccessServer from OPC. 

It fails in the creation of UaTcpSessionClient

`// Create the session client for the app.
            this.session = new UaTcpSessionClient(
                new ApplicationDescription()
                {
                    ApplicationName = "Workstation.MobileHmi",
                    ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:Workstation.MobileHmi",
                    ApplicationType = ApplicationType.Client
                },
                new DirectoryStore(
                    Path.Combine(System.Environment.GetFolderPath(System.Environment.SpecialFolder.LocalApplicationData), "pki"),
                    loggerFactory: loggerFactory),
                this.ProvideUserIdentity,
                this.discoveryUrl,
                this.loggerFactory);`

awcullen, 2017-01-03:
For my test on the Samsung phone, the phone was connected by wifi to my home network. I starting the UA C++ Demo Server on my desktop. The phone connected to the desktop using url: opc.tcp://192.168.1.108:48010.

For the OPC Foundation's DA Server, I would try url opc.tcp://[ip address]:62547

CarlCrol, 2017-01-03:
I tested two servers. One is auto-accept and one is from the foundation test servers.

I was not able to see the rejected certificate from the foundation test servers which i think that my test client appliction didn't connect successfully to the foundation test servers.

awcullen, 2017-01-03:
Could you copy the debug text from the output window and put it in a file on DropBox?
Thanks.

CarlCrol, 2017-01-06:
The IP Address is accessible since the test application i created using Prosys SDK in Java works (trial version).
I also tried some Ping tools and it is accessible.

> Android application is debugging.
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/App1.Droid.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/BouncyCastle.Crypto.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/FormsViewGroup.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Microsoft.Extensions.DependencyInjection.Abstractions.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Microsoft.Extensions.Logging.Abstractions.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Microsoft.Extensions.Logging.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Threading.Tasks.Dataflow.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Workstation.UaClient.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Xamarin.Android.Support.Design.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Xamarin.Android.Support.v4.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Xamarin.Android.Support.v7.AppCompat.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Xamarin.Android.Support.v7.CardView.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Xamarin.Android.Support.v7.MediaRouter.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Xamarin.Forms.Core.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Xamarin.Forms.Platform.Android.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Xamarin.Forms.Platform.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Xamarin.Forms.Xaml.dll
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/Java.Interop.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Runtime.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Resources.ResourceManager.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Diagnostics.Debug.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.IO.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Globalization.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Runtime.Extensions.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.IO.FileSystem.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Collections.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Reflection.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Reflection.Extensions.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Linq.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Threading.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Text.Encoding.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.IO.FileSystem.Primitives.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Threading.Tasks.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.ComponentModel.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Linq.Expressions.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Collections.Concurrent.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Dynamic.Runtime.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Diagnostics.Tracing.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Xml.XDocument.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.ObjectModel.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Security.Cryptography.X509Certificates.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Net.Sockets.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Text.Encoding.Extensions.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Xml.ReaderWriter.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.ServiceModel.Internals.dll [External]
[... cut ...]
01-06 14:40:04.601 D/ViewRootImpl(28632): #1 mView = com.android.internal.policy.PhoneWindow$DecorView{68ffd70 I.E...... R.....ID 0,0-0,0}
01-06 14:40:04.661 I/InjectionManager(28632): dispatchCreateOptionsMenu :md50170a25da682dd459b9251c65ebe3a7a.MainActivity
01-06 14:40:04.671 D/Mono    (28632): Assembly Ref addref Xamarin.Forms.Platform.Android[0xe9992ec0] -> Xamarin.Android.Support.v4[0xe9992da0]: 2
Unhandled Exception:

System.NullReferenceException: Object reference not set to an instance of an object.

CarlCrol, 2017-01-10:
any updates on the issue?

awcullen, 2017-01-10:
[image]
[image]

I have your App1 working on my Samsung S6 running Android 6.0.1.
Here's how I did it. 

On the phone I activated Developer Mode and made sure USB Debugging was enabled. I connected the phone to my notebook via the USB cable. Visual Studio discovered the device and added it as a target for debugging. I ensured that the phone also had a WIFI connection to my home network. 

I made these changes to your App1.Droid project :
Added a Nuget package: Microsoft.Extensions.Logging.Debug v1.0.1
In MainActivity.cs, I changed the discoveryUrl to use the IP Address of my notebook. I added a line to cause the logger to write to the debugger.
```
   private string discoveryUrl = @"opc.tcp://192.168.1.109:62547/Quickstarts/DataAccessServer";
     // Setup a logger.
    this.loggerFactory = new LoggerFactory();
    this.loggerFactory.AddDebug(LogLevel.Trace); // added this line
    this.logger = this.loggerFactory.CreateLogger<MainActivity>();

```
On the notebook:
I started the OPC Foundation Data Access Server. I then started debugging App1 in Visual Studio. The selected target was the Samsung SM-G920A.
Initially the debugger repeated:
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel opening.
[0:] Workstation.ServiceModel.Ua.DirectoryStore: Trace: Found certificate with subject alt name 'urn:localhost:Workstation.MobileHmi'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: Opening secure channel with endpoint 'opc.tcp://192.168.1.109:62547'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityPolicy: 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityMode: 'SignAndEncrypt'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: UserIdentityToken: 'Workstation.ServiceModel.Ua.AnonymousIdentity'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending OpenSecureChannelRequest Handle: 2
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Canceled OpenSecureChannelRequest Handle: 2
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel faulted.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel closing.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Waiting for socket to close.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error opening channel with endpoint 'opc.tcp://192.168.1.109:62547'. Timeout occurred while processing the request.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: State machine retrying. Timeout occurred while processing the request.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Error: Timeout while waiting for socket to close.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel closed.

I opened the folder:
C:\ProgramData\OPC Foundation\CertificateStores\RejectedCertificates\certs
I moved the .der file from the above directory to the following:
C:\ProgramData\OPC Foundation\CertificateStores\UA Applications\certs

The next connection retry was successful.

CarlCrol, 2017-01-17:
thanks will try this as soon as possible.

CarlCrol, 2017-01-17:
I tried pinging the machine where the server is running through my phone. I was able to do it. But still the exception still occurs with your suggestion.

01-17 09:15:43.961 D/Mono    ( 7649): Assembly Loader probing location: '/storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.IO.dll'.
01-17 09:15:43.961 D/Mono    ( 7649): AOT module '/storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.IO.dll.so' not found: dlopen failed: library "/data/app/App1.Droid-1/lib/arm/libaot-System.IO.dll.so" not found
01-17 09:15:43.961 D/Mono    ( 7649): AOT module '/Users/builder/data/lanes/3415/7db2aac3/source/monodroid/builds/install/mono-armv7/lib/mono/aot-cache/arm/System.IO.dll.so' not found: dlopen failed: library "/data/app/App1.Droid-1/lib/arm/libaot-System.IO.dll.so" not found
01-17 09:15:43.961 D/Mono    ( 7649): Assembly Ref addref BouncyCastle.Crypto[0xeb183680] -> System.IO[0xeb184100]: 3
01-17 09:15:43.981 D/Mono    ( 7649): Assembly Loader probing location: '/storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Globalization.dll'.
01-17 09:15:43.981 D/Mono    ( 7649): AOT module '/storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Globalization.dll.so' not found: dlopen failed: library "/data/app/App1.Droid-1/lib/arm/libaot-System.Globalization.dll.so" not found
01-17 09:15:43.981 D/Mono    ( 7649): AOT module '/Users/builder/data/lanes/3415/7db2aac3/source/monodroid/builds/install/mono-armv7/lib/mono/aot-cache/arm/System.Globalization.dll.so' not found: dlopen failed: library "/data/app/App1.Droid-1/lib/arm/libaot-System.Globalization.dll.so" not found
01-17 09:15:43.981 D/Mono    ( 7649): Assembly Ref addref BouncyCastle.Crypto[0xeb183680] -> System.Globalization[0xeb184520]: 2
01-17 09:15:44.011 D/Mono    ( 7649): Assembly Loader probing location: '/storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Collections.dll'.
01-17 09:15:44.011 D/Mono    ( 7649): AOT module '/storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Collections.dll.so' not found: dlopen failed: library "/data/app/App1.Droid-1/lib/arm/libaot-System.Collections.dll.so" not found
01-17 09:15:44.011 D/Mono    ( 7649): AOT module '/Users/builder/data/lanes/3415/7db2aac3/source/monodroid/builds/install/mono-armv7/lib/mono/aot-cache/arm/System.Collections.dll.so' not found: dlopen failed: library "/data/app/App1.Droid-1/lib/arm/libaot-System.Collections.dll.so" not found
01-17 09:15:44.011 D/Mono    ( 7649): Assembly Ref addref BouncyCastle.Crypto[0xeb183680] -> System.Collections[0xeb184880]: 6
01-17 09:15:44.021 D/Mono    ( 7649): Assembly Loader probing location: '/storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Reflection.dll'.
01-17 09:15:44.021 D/Mono    ( 7649): AOT module '/storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Reflection.dll.so' not found: dlopen failed: library "/data/app/App1.Droid-1/lib/arm/libaot-System.Reflection.dll.so" not found
01-17 09:15:44.021 D/Mono    ( 7649): AOT module '/Users/builder/data/lanes/3415/7db2aac3/source/monodroid/builds/install/mono-armv7/lib/mono/aot-cache/arm/System.Reflection.dll.so' not found: dlopen failed: library "/data/app/App1.Droid-1/lib/arm/libaot-System.Reflection.dll.so" not found
01-17 09:15:44.021 D/Mono    ( 7649): Assembly Ref addref BouncyCastle.Crypto[0xeb183680] -> System.Reflection[0xeb184940]: 5
01-17 09:15:44.041 D/Mono    ( 7649): Assembly Loader probing location: '/storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Threading.dll'.
01-17 09:15:44.041 D/Mono    ( 7649): AOT module '/storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Threading.dll.so' not found: dlopen failed: library "/data/app/App1.Droid-1/lib/arm/libaot-System.Threading.dll.so" not found
01-17 09:15:44.041 D/Mono    ( 7649): AOT module '/Users/builder/data/lanes/3415/7db2aac3/source/monodroid/builds/install/mono-armv7/lib/mono/aot-cache/arm/System.Threading.dll.so' not found: dlopen failed: library "/data/app/App1.Droid-1/lib/arm/libaot-System.Threading.dll.so" not found
01-17 09:15:44.041 D/Mono    ( 7649): Assembly Ref addref BouncyCastle.Crypto[0xeb183680] -> System.Threading[0xeb184a60]: 6
01-17 09:15:44.101 D/Mono    ( 7649): DllImport attempting to load: '__Internal'.
01-17 09:15:44.101 D/Mono    ( 7649): DllImport loaded library '(null)'.
01-17 09:15:44.101 D/Mono    ( 7649): DllImport searching in: '__Internal' ('(null)').
01-17 09:15:44.101 D/Mono    ( 7649): Searching for 'monodroid_get_system_property'.
01-17 09:15:44.101 D/Mono    ( 7649): Probing 'monodroid_get_system_property'.
01-17 09:15:44.101 D/Mono    ( 7649): Found as 'monodroid_get_system_property'.
01-17 09:15:44.101 D/Mono    ( 7649): DllImport searching in: '__Internal' ('(null)').
01-17 09:15:44.101 D/Mono    ( 7649): Searching for 'monodroid_free'.
01-17 09:15:44.101 D/Mono    ( 7649): Probing 'monodroid_free'.
01-17 09:15:44.101 D/Mono    ( 7649): Found as 'monodroid_free'.
01-17 09:15:44.181 D/Mono    ( 7649): Assembly Loader probing location: '/storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Threading.Tasks.dll'.
01-17 09:15:44.181 D/Mono    ( 7649): AOT module '/storage/emulated/0/Android/data/App1.Droid/files/.__override__/System.Threading.Tasks.dll.so' not found: dlopen failed: library "/data/app/App1.Droid-1/lib/arm/libaot-System.Threading.Tasks.dll.so" not found
[... cut ...]
01-17 09:15:44.381 W/Mono    ( 7649): The class System.Runtime.CompilerServices.ConditionalWeakTable`2 could not be loaded, used in System.Runtime, Version=4.0.20.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a
01-17 09:15:44.381 W/Mono    ( 7649): The class System.Runtime.CompilerServices.ConditionalWeakTable`2 could not be loaded, used in System.Runtime, Version=4.0.20.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a
01-17 09:16:10.501 D/Mono    ( 7649): [0xd81cd930] worker finishing
Thread finished: <Thread Pool> #6

awcullen, 2017-01-25:
Here are some things to check..
[image]

[image]

[image]

closed 2017-01-29 by awcullen