convertersystems/opc-ua-samples #7: Workstation.MobileHmi.Droid project compilation errors

abrasat, 2016-12-22:
I am trying to compile the new version but am getting lots of errors like:

...
Please install package: 'Xamarin.Android.Support.v7.CardView' available in SDK installer. Android resource directory C:\Users\***\AppData\Local\Xamarin\Xamarin.Android.Support.v7.CardView\23.3.0.0\content\./ doesn't exist.
...
Please install package: 'Xamarin.Android.Support.Vector.Drawable' available in SDK installer. Java library file C:\Users\***\AppData\Local\Xamarin\Xamarin.Android.Support.Vector.Drawable\23.3.0.0\content\classes.jar doesn't exist.
...
Please install package: 'Xamarin.Android.Support.v4' available in SDK installer. Android resource directory C:\Users\***\AppData\Local\Xamarin\Xamarin.Android.Support.v4\23.3.0.0\content\./ doesn't exist.

... and so on.

I installed these packages manually from the nuget server, but the error messages are still generated ...
Why are these packages reported as missing, respectively why are they searched into the c:\users\... folder ?

awcullen, 2016-12-22:
I am just learning about Xamarin as well.
To get the SDK for Xamarin Android support, I chose Tools...Android...Android SDK Manager.
Then I installed the following items: 
[image]

awcullen, 2016-12-22:
Next I created an Android Virtual Device (emulator).
I chose Tools...Android...Android Emulator Manager:
[image]

I selected the Nexus 4 by Google and clicked Create AVD...
 
I filled out the form as shown:
[image]

abrasat, 2017-01-02:
After downlaoding the OPC-UA samples and trying to compile it with VS 2015, the following errors occur:
...
The package Microsoft.CSharp with version 4.0.1 could not be found in C:\Users\Andrew\.nuget\packages\. Run a NuGet package restore to download the package.	Workstation.Common.Uwp			
...
Your project is not referencing the "MonoAndroid,Version=v6.0" framework. Add a reference to "MonoAndroid,Version=v6.0" in the "frameworks" section of your project.json, and then re-run NuGet restore.	Workstation.MobileHmi.Droid			
...

The project.json file in the "Workstation.Common.Uwp" project seems to contain some references to directories from your development environment ("C:\Users\Andrew\...").

In the project.json file from the "Workstation.MobileHmi.Droid" project seems that some references (MonoAndroid) are missing.

awcullen, 2017-01-06:
To help fix the first part of the question:
I removed the project.lock.json from the repo. This file is recreated if necessary when the project is rebuilt.
I lowerer the target Android API to 23 and MonoAndroid to v6.0, which are more commonly installed. 

Regarding the last Xamarin Forms error:
Please check the properties of the MainPage.xaml in the shared project. Then restart Visual Studio.
[image]

abrasat, 2017-01-08:
Thanks, I will look into that, but probably first in a couple of days.
Do you have the RobotHmi sample also as a Xamarin/Android version ? If yes, could you please add it to the repository of samples?

awcullen, 2017-01-29:
I'm working on it. https://urho3d.github.io/

closed 2017-01-29 by awcullen