convertersystems/opc-ua-samples #10: Error when I have a newer Version of core dlls...

jogibear9988, 2017-01-28:
I've a Newer Version of Microsoft.Extensions.Logging.Abstractions in a Net4.5 Project. But I get a Exception...

awcullen, 2017-01-28:
Try 1.5.5 for me.

jogibear9988, 2017-01-28:
Same Exception again... 
It works if I add this to app.config

    <runtime>
        <assemblyBinding xmlns="urn:schemas-microsoft-com:asm.v1">
            <dependentAssembly>
                <assemblyIdentity name="Microsoft.Win32.Primitives" publicKeyToken="b03f5f7f11d50a3a" culture="neutral" />
                <bindingRedirect oldVersion="0.0.0.0-99.9.9.9" newVersion="4.0.1.0" />
            </dependentAssembly>
            <dependentAssembly>
                <assemblyIdentity name="System.IO.Compression" publicKeyToken="b77a5c561934e089" culture="neutral" />
                <bindingRedirect oldVersion="0.0.0.0-99.9.9.9" newVersion="4.1.0.0" />
            </dependentAssembly>
            <dependentAssembly>
                <assemblyIdentity name="System.Security.Cryptography.X509Certificates" publicKeyToken="b03f5f7f11d50a3a" culture="neutral" />
                <bindingRedirect oldVersion="0.0.0.0-99.9.9.9" newVersion="4.0.0.0" />
            </dependentAssembly>
            <dependentAssembly>
                <assemblyIdentity name="System.Security.Cryptography.Algorithms" publicKeyToken="b03f5f7f11d50a3a" culture="neutral" />
                <bindingRedirect oldVersion="0.0.0.0-99.9.9.9" newVersion="4.0.0.0" />
            </dependentAssembly>
            <dependentAssembly>
                <assemblyIdentity name="Microsoft.Extensions.Logging.Abstractions" publicKeyToken="adb9793829ddae60" culture="neutral" />
                <bindingRedirect oldVersion="0.0.0.0-99.9.9.9" newVersion="1.1.0.0" />
            </dependentAssembly>
        </assemblyBinding>
    </runtime>

but how will I be able to fix in the feature, when I include different assemblys, where they targeted different Logging dlls during build time?

jogibear9988, 2017-01-28:
And on my build server I got now this error:

[ResolveAssemblyReference] C:\Windows\Microsoft.NET\Framework\v4.0.30319\Microsoft.Common.targets(1605, 5): warning MSB3258: Der Primärverweis "E:\BuildAgent\work\5f3ffb1cd33a89d9\MCC.V1.0\bin\Debug\MCC.Common.ServiceInterfaces.dll" konnte nicht aufgelöst werden, da er eine indirekte Abhängigkeit von der .NET Framework-Assembly "System.IO.Compression, Version=4.1.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089" aufweist, die eine höhere Version 4.1.0.0 als die Version 4.0.0.0 im aktuellen Zielframework aufweist.
[11:00:44]

jogibear9988, 2017-01-28:
seams .net 4.5 only supports 4.0.0.0 of io.compression? but why does it work on my machine? (the whole solution targets net 4.5)

closed 2017-03-23 by awcullen