convertersystems/opc-ua-samples #11: Can we connect this data to Azure and live data show on azure?

tejasklodha, 2017-02-13:
Dear Sir,

It is not any issue but some new idea.....
Can we connect this UWP app to azure and live data show on azure?

As per https://github.com/Azure/iot-gateway-opc-ua link we can connect OPC UZ to Azure so can you demonstrate how we can do?

Also I refer below link..
https://blogs.microsoft.com/iot/2016/11/22/microsoft-continues-azure-iot-suite-integration-with-opc-ua-and-partners-at-sps-ipc-drives-2016/

Thanks,

awcullen, 2017-02-22:
This is an excellent idea for a new sample!

tejasklodha, 2017-02-26:
yup,

I found some interesting
 
Please see here for the Azure-hosted other end of the solution:
https://github.com/OPCFoundation/UA-.NETStandardLibrary/tree/master/OpcUaWebTelemetry

Can you connect your this app with azure in uwp?

tejasklodha, 2017-03-11:
Can you demonstrate to connect wid cloud?

tejasklodha, 2017-05-01:
Done...... 👍

closed 2017-05-01 by tejasklodha