convertersystems/opc-ua-samples #12: Problem to  Load UWP app to Raspberry Pi

tejasklodha, 2017-02-14:
Dear Sir,

When I am trying to load Workstation.RobotApp UWP app but not able to load raspberry pi and give below exception

An exception of type 'System.BadImageFormatException' occurred in Urho.dll but was not handled in user code
Additional information: An attempt was made to load a program with an incorrect format. (Exception from HRESULT: 0x8007000B)

Exception occur on below line in "RobotControl.xaml.cs"
 this.robotGame = this.UrhoSurface.Run<RobotGame>(new ApplicationOptions("Assets") { Width = (int)this.UrhoSurface.ActualWidth, Height = (int)this.UrhoSurface.ActualHeight });

Tejas Lodha

awcullen, 2017-02-14:
Reading some posts on Windows Core IOT leads me to think that hardware accelerated graphics is not yet supported on RPi . Maybe on Minnowboard or Dragonboard, but i'm not sure.

tejasklodha, 2017-03-08:
It most helpful if your UWP code support Windows Core IOT....
I am waiting from Microsoft when they start to give support hardware accelerated graphics on RPi...

Thanks for sharing your Awesome knowledge to US

Thanks,
Tejas Lodha

closed 2017-03-08 by tejasklodha