convertersystems/opc-ua-samples #13: documentations about the library

roozbeh63, 2017-02-21:
where I can find more documentations about using this library, your example are already good, but I need a proper documentations to review. 
thanks

awcullen, 2017-02-21:
There is no documentation for this library yet.
You may find the [specifications](https://opcfoundation.org/developer-tools/specifications-unified-architecture) of OPC UA interesting. I tried to follow these as best I could.

roozbeh63, 2017-02-24:
if it is not too much to ask and you have time, can you please create a windows form example of your library? with simple functions, just a simple read and write values to some nodes. 
much thanks in advance

awcullen, 2019-07-28:
Sorry, we won't be supporting WinForms.

closed 2019-07-28 by awcullen