convertersystems/opc-ua-samples #14: MobileHmi.Uwp ?

abrasat, 2017-02-21:
Would be possible to get the MobileHmi sample working also on UWP ?
As I did not use UrhoSharp by now, could you please tell me what tools are you using to design the UrhoSharp 3D graphical objects ?

awcullen, 2017-02-21:
I think it would be quite possible to have the MobileHmi sample working on UWP and iOS. It's just a matter of time.

Regarding the 3d assets for UrhoSharp, I used the open-source tool 'Blender' as well as the add-in 'Blender to Urho Exporter'.

abrasat, 2017-02-22:
Thanks for the information.
Is it possible to bind the node identifier string (the attribute from the view model element) to a graphic item (for instance a label) ?

awcullen, 2017-02-22:
Sorry there no support in XAML for binding to Attributes

tejasklodha, 2017-02-26:
Hi awcullen,

Are U used https://github.com/reattiva/Urho3D-Blender ?

awcullen, 2017-02-27:
Yes, thanks for finding the link. I corrected my previous comment.

abrasat, 2017-02-27:
I know that it is not directly related to opc-ua, but do you know if it is possible also to import UrhoSharp 3d assets back to Blender, in view to edit them ? Or how do you modify them ?
Can you recommend some good sites with 3d models for Blender/UrhoSharp ?

awcullen, 2017-02-28:
I don't know of a way to get the Urho3d scene assets back into Blender. 
Google is a good tool to find 3d models. Blender imports so many formats.

abrasat, 2017-03-15:
In the MobileHmi sample the Fps is limited by setting the MinFps and MaxFps engine parameters. How can the current Fps be determined ?

closed 2017-03-23 by awcullen