convertersystems/opc-ua-samples #16: Publish several subscriptions

abrasat, 2017-02-27:
Could you please post a sample about how to publish several subscriptions ?
In the ConsoleApp sample the PublishRequest class is used as follows

> var publishRequest = new PublishRequest
    {
        SubscriptionAcknowledgements = new SubscriptionAcknowledgement[0]
     };
...
var publishResponse = await session.PublishAsync(publishRequest).ConfigureAwait(false);

How does the PublishRequest relate to the previously created Subscription/MonitoringItemRequest ? What if there are several subcriptions ?

awcullen, 2017-02-27:
When you call PublishAsync, the response can be for any subscription of the session. PublishResponse has field for SubscriptionId, which will match a SubscriptionId that was previously returned in the CreateSubscriptionResponse.
   
I edited 'program.cs', adding Subscription ID to the console output to be more clear.

abrasat, 2017-02-27:
Thanks for that.
But what about the first "dummy" PublishRequest object before the while loop (at this point there is no publish response available yet) , it does not contain any subscription id. The first call to PublishAsync() in your sample uses that "dummy" publish request, and only the subsequent calls use the subscription id from the PublishResponse.
Also it would be helpful if you could extend the sample with the usage of 2 different subscriptions for the  same session.

> Console.WriteLine("Step 8 - Publish the subscription.");
var publishRequest = new PublishRequest
    {
            SubscriptionAcknowledgements = new SubscriptionAcknowledgement[0]
     };
     Console.WriteLine("Press any key to delete the subscription...");
     while (!Console.KeyAvailable)
     {
              var publishResponse = await session.PublishAsync(publishRequest).ConfigureAwait(false);
              // loop thru all the data change notifications
              var dcns = publishResponse.NotificationMessage.NotificationData.OfType<DataChangeNotification>);
....

awcullen, 2017-02-27:
I see the confusion. The SubscriptionAcknowledgements field holds the sequence number of the _previous_ PublishResponse. The purpose is to allow the server to release the memory of the _previous_ PublishResponse (Servers have to keep old responses around until they are Acknowledged)

closed 2017-03-23 by awcullen