convertersystems/opc-ua-samples #19: Certificate Error in Workstation.ConsoleApp and also in Workstation.DataLoggingConsole

tejasklodha, 2017-03-08:
Dear Sir,

I am getting error while run Workstation.ConsoleApp and Workstation.DataLoggingConsole Project as below in command prompt 

"Step 1 - Describe this app.
Step 2 - Discover endpoints of 'opc.tcp://localhost:26543'.
Step 3 - Choose the endpoint with highest security level.
http://opcfoundation.org/UA/SecurityPolicy#Basic256
Step 4 - Create a session with your server.
The Certificate has expired or is not yet valid.
Press any key to close the program..."

in same project I am able to successfully run Workstation.RobotApp(UWP) but I am not able to run Workstation.RobotHmi and Error is Timeout

I think issue is Certificate location Define in code.....
Need your help

Tejas Lodha

awcullen, 2017-03-09:
Look in program.cs and you will find:
```
             Console.WriteLine("Step 4 - Create a session with your server.");
            var session = new UaTcpSessionChannel(
                appDescription,
                new DirectoryStore(Environment.ExpandEnvironmentVariables(@"%LOCALAPPDATA%\Workstation.ConsoleApp\pki")),
                userIdentity,
                remoteEndpoint);
```
With explorer you can browse to this directory:
%LOCALAPPDATA%\Workstation.ConsoleApp\pki

Open the "\own" directory and you will find the "client.crt". If you open the "crt" file it will display "Valid from x to y." If the dates are wrong you can delete the "client.crt" and the program will create a new cert on the next run..

tejasklodha, 2017-03-10:
Yup.....

Code working fine wn delete the file
Thanks for your quick response.....

closed 2017-03-10 by tejasklodha