convertersystems/opc-ua-samples #20: Connecting to RobotServer

abrasat, 2017-03-22:
The MobileHmi.Droid and RobotHmi client samples do not connect to the new RobotServer.
Both samples work fine with the "old" node-based server form opc-ua-samples.
Are some client/server new certificates to be created ? If yes, how ?

awcullen, 2017-03-22:
Please check that the file Workstation.RobotServer.Config.xml has at line 41
<AutoAcceptUntrustedCertificates>true</AutoAcceptUntrustedCertificates>

Run RobotHmi and paste the contents of the Debug Window here, that might help me.

The license of the OPC UA .Net Standard Library SDK used in the RobotServer remains GPL 2.0

abrasat, 2017-03-22:
The RobotServer Setting is on true.

The RobotHmi started working now, after I connected it to the NodeServer (maybe is just a coincidence)

 MobileHmi.Droid does not work, this is the output from. Is the discovery port in RobotServer also 26543 ?

...
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error discovering endpoints of 'opc.tcp://aa.bb.cc.dd:26543'. Connection timed out
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Information: Closing channel with endpoint 'opc.tcp://aa.bb.cc.dd:26543'.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error closing channel. Object reference not set to an instance of an object.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Trace: Connecting in 2000 ms.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Information: Discovering endpoints of 'opc.tcp://aa.bb.cc.dd::26543'.

With the same settings MobileHmi.Droid connects to NodeServer.

awcullen, 2017-03-22:
I'm running it on the Android Emulator for x86 (Android 6.0 - API23) and its working normally.

awcullen, 2017-03-22:
Remember when running on an emulator you must use the ip address ( or '10.0.2.2' for accessing local host computer from emulator.  Hostname, localhost or 127.0.0.1 will not work .

```private string discoveryUrl = @"opc.tcp://10.0.2.2:26543"; ```

tejasklodha, 2017-03-22:
I m also able to run Android app in my  motorola mobile.....

It is working fine....

abrasat, 2017-03-23:
I changed the port on the RobotServer and now its working fine. Somehow the port 26543 was "cached" as assigned to the NodeServer.
Is it possible to configure the timeout in opc-ua-client, until it gives up trying to connect to a (not available) opc-ua server ?
Are you planning a opc-ua-server component in the future ?

closed 2017-03-23 by awcullen