convertersystems/opc-ua-samples #21: RobotApp disconnects from a opc server after window being minimized

rudoc, 2017-03-28:
Dear Andrew, thanks for the great example of OPC UA application!
I am using UWP RobotApp part of your project with connection to Codesys RTE.
I have found that in couple of seconds after being minimized RobotApp client disconnects from a server. When running in full screen or window mode connection remains stable.
I suspect that this is drawback from v1.4.1 release.
Could you please confirm if this behavior is common or i should be checking my setup?

rudoc, 2017-03-28:
I was able to reproduce the behavior of the RobotApp  application from current project:
1.Deploy the RobotApp (Debug mode)
2. Run RobotServer
3. Run RobotApp
4. Wait for connection with RobotServer
5. Minimize RobotApp window… - disconnection in 2-3 seconds.
When I started both projects from VS IDE, RobotApp stayed connected even being minimized.
Win10x64 +VS2017.

awcullen, 2017-03-28:
I not sure if I agree with Microsoft on this one, but a App is placed in Suspended mode when minimized.
See the first part of this blog post: [the-lifecycle-of-a-uwp-app](https://blogs.windows.com/buildingapps/2016/04/28/the-lifecycle-of-a-uwp-app)

In the app.xaml.cs you can try removing this code:
```
public async override Task OnSuspendingAsync(object s, SuspendingEventArgs e, bool prelaunchActivated)
{
    await this.container.Resolve<PLC1Session>().SuspendAsync();
}

public override void OnResuming(object s, object e, AppExecutionState previousExecutionState)
{
    this.container.Resolve<PLC1Session>().Resume();
}

```

rudoc, 2017-03-29:
Andrew!
Thanks for the pointing the disconnection reason. Unfortunately, removing the code above did not help. I will think on workaround for this.

closed 2017-03-29 by rudoc