convertersystems/opc-ua-samples #23: The message encoding/decoding limits imposed by the stack have been exceeded.

Savoyyang, 2017-04-21:
Hello ,awcullen

I use your UaAndroid to connect PROSYS Simulation Server,the following error occurred，
I found this [link](http://forum.prosysopc.com/forum/opc-ua-java-sdk/serviceexception-bad_encodinglimitsexceeded/) talk about my error,you  can look at what's the problem.

Thank for you contribution,
Best regards.

Workstation.ServiceModel.Ua.UaTcpSessionClient: Information: Discovering endpoints of 'opc.tcp://ip:53530'.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error discovering endpoints of 'opc.tcp://ip:53530'. The message encoding/decoding limits imposed by the stack have been exceeded.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Information: Closing channel with endpoint 'opc.tcp://ip:53530'.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error closing channel. Object reference not set to an instance of an object.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Trace: Connecting in 20000 ms.

awcullen, 2017-04-21:
Thank you for asking me to test Prosys UA Simulation Server. It's very nice.

I found my bug that caused this ServiceResultException. The Prosys server returns MaxMessageSize = 0, which should mean no limit.

But another detail was even more tricky. Prosys wants you to use the complete, full endpointUrl when you connect: 'opc.tcp://ip:53530/OPCUA/SimulationServer' 

Please update your NuGet package to 1.5.12

Savoyyang, 2017-04-24:
I test the full endpointUrl and update NuGet package to 1.5.12 ,but still Unable to connect .

Savoyyang, 2017-04-24:
I test the full endpointUrl and update NuGet package to 1.5.12 ,but still Unable to connect .

[image]

[image]

awcullen, 2017-04-24:
First, please check if the certificate is trusted by the Simulation Server. The server has a Certificates tab. Find this android app's certificate and if it says "rejected" then right-click on it and select "trust".
 
If that was not the problem then you should find some messages in the Debug Output that contain "Workstation.ServiceModel.Ua".  Post them here.

Savoyyang, 2017-04-25:
Thread started:  #7
InspectorDebugSession(4): HandleTargetEvent: ThreadStarted
04-25 09:14:16.439 D/Mono    ( 3267): Image addref System.Text.Encoding[0xb8e79f80] -> System.Text.Encoding.dll[0xb8e72040]: 2
04-25 09:14:16.439 D/Mono    ( 3267): Prepared to set up assembly 'System.Text.Encoding' (System.Text.Encoding.dll)
04-25 09:14:16.439 D/Mono    ( 3267): Assembly System.Text.Encoding[0xb8e79f80] added to domain RootDomain, ref_count=1
04-25 09:14:16.439 D/Mono    ( 3267): AOT module 'System.Text.Encoding.dll.so' not found: dlopen failed: library "/data/app-lib/Workstation.MobileHmi-2/libaot-System.Text.Encoding.dll.so" not found
04-25 09:14:16.439 D/Mono    ( 3267): AOT module '/Users/builder/data/lanes/4468/f913a78a/source/monodroid/builds/install/mono-x86/lib/mono/aot-cache/x86/System.Text.Encoding.dll.so' not found: dlopen failed: library "/data/app-lib/Workstation.MobileHmi-2/libaot-System.Text.Encoding.dll.so" not found
04-25 09:14:16.439 D/Mono    ( 3267): Config attempting to parse: 'System.Text.Encoding.dll.config'.
04-25 09:14:16.439 D/Mono    ( 3267): Config attempting to parse: '/Users/builder/data/lanes/4468/f913a78a/source/monodroid/builds/install/mono-x86/etc/mono/assemblies/System.Text.Encoding/System.Text.Encoding.config'.
04-25 09:14:16.439 D/Mono    ( 3267): Assembly Ref addref Workstation.UaClient[0xb7920300] -> System.Text.Encoding[0xb8e79f80]: 2
04-25 09:14:16.439 D/Mono    ( 3267): Assembly Ref addref System.Text.Encoding[0xb8e79f80] -> mscorlib[0xb78d47c0]: 44
Loaded assembly: System.Text.Encoding.dll [External]
04-25 09:14:16.459 D/Mono    ( 3267): DllImport searching in: 'mono-urho' ('/storage/emulated/0/Android/data/Workstation.MobileHmi/files/.__override__/libmono-urho.so').
04-25 09:14:16.459 D/Mono    ( 3267): Searching for 'Context_Context'.
04-25 09:14:16.459 D/Mono    ( 3267): Probing 'Context_Context'.
04-25 09:14:16.459 D/Mono    ( 3267): Found as 'Context_Context'.
04-25 09:14:16.463 D/Mono    ( 3267): Assembly Ref addref Urho.Forms[0xb791f4d0] -> System.Core[0xb7b4a270]: 8
04-25 09:14:16.475 D/Mono    ( 3267): DllImport searching in: 'mono-urho' ('/storage/emulated/0/Android/data/Workstation.MobileHmi/files/.__override__/libmono-urho.so').
04-25 09:14:16.475 D/Mono    ( 3267): Searching for 'RefCounted_Refs'.
04-25 09:14:16.475 D/Mono    ( 3267): Probing 'RefCounted_Refs'.
04-25 09:14:16.475 D/Mono    ( 3267): Found as 'RefCounted_Refs'.
04-25 09:14:16.491 D/Mono    ( 3267): DllImport searching in: 'mono-urho' ('/storage/emulated/0/Android/data/Workstation.MobileHmi/files/.__override__/libmono-urho.so').
04-25 09:14:16.491 D/Mono    ( 3267): Searching for 'ApplicationProxy_ApplicationProxy'.
04-25 09:14:16.491 D/Mono    ( 3267): Probing 'ApplicationProxy_ApplicationProxy'.
04-25 09:14:16.491 D/Mono    ( 3267): Found as 'ApplicationProxy_ApplicationProxy'.
04-25 09:14:16.527 D/Mono    ( 3267): DllImport searching in: 'mono-urho' ('/storage/emulated/0/Android/data/Workstation.MobileHmi/files/.__override__/libmono-urho.so').
04-25 09:14:16.527 D/Mono    ( 3267): Searching for 'Application_Run'.
04-25 09:14:16.527 D/Mono    ( 3267): Probing 'Application_Run'.
04-25 09:14:16.527 D/Mono    ( 3267): Found as 'Application_Run'.
04-25 09:14:16.527 D/Mono    ( 3267): DllImport searching in: 'mono-urho' ('/storage/emulated/0/Android/data/Workstation.MobileHmi/files/.__override__/libmono-urho.so').
04-25 09:14:16.527 D/Mono    ( 3267): Searching for 'UrhoObject_GetTypeName'.
04-25 09:14:16.527 D/Mono    ( 3267): Probing 'UrhoObject_GetTypeName'.
04-25 09:14:16.527 D/Mono    ( 3267): Found as 'UrhoObject_GetTypeName'.
04-25 09:14:16.527 D/Mono    ( 3267): DllImport searching in: 'mono-urho' ('/storage/emulated/0/Android/data/Workstation.MobileHmi/files/.__override__/libmono-urho.so').
04-25 09:14:16.527 D/Mono    ( 3267): Searching for 'UrhoObject_GetContext'.
04-25 09:14:16.527 D/Mono    ( 3267): Probing 'UrhoObject_GetContext'.
04-25 09:14:16.527 D/Mono    ( 3267): Found as 'UrhoObject_GetContext'.
04-25 09:14:16.567 I/Urho3D  ( 3267): Created 1 worker thread
Thread started: <Thread Pool> #8
04-25 09:14:16.579 I/Urho3D  ( 3267): Added resource package /apk/CoreData.pak
InspectorDebugSession(4): HandleTargetEvent: ThreadStarted
04-25 09:14:16.623 D/Mono    ( 3267): Image addref System.Text.Encoding.Extensions[0xb8f9f800] -> System.Text.Encoding.Extensions.dll[0xb8f90330]: 2
04-25 09:14:16.623 D/Mono    ( 3267): Prepared to set up assembly 'System.Text.Encoding.Extensions' (System.Text.Encoding.Extensions.dll)
04-25 09:14:16.623 D/Mono    ( 3267): Assembly System.Text.Encoding.Extensions[0xb8f9f800] added to domain RootDomain, ref_count=1
04-25 09:14:16.623 D/Mono    ( 3267): AOT module 'System.Text.Encoding.Extensions.dll.so' not found: dlopen failed: library "/data/app-lib/Workstation.MobileHmi-2/libaot-System.Text.Encoding.Extensions.dll.so" not found
04-25 09:14:16.631 D/Mono    ( 3267): AOT module '/Users/builder/data/lanes/4468/f913a78a/source/monodroid/builds/install/mono-x86/lib/mono/aot-cache/x86/System.Text.Encoding.Extensions.dll.so' not found: dlopen failed: library "/data/app-lib/Workstation.MobileHmi-2/libaot-System.Text.Encoding.Extensions.dll.so" not found
04-25 09:14:16.631 D/Mono    ( 3267): Config attempting to parse: 'System.Text.Encoding.Extensions.dll.config'.
[... cut ...]
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error discovering endpoints of 'opc.tcp://192.168.0.186:53530/OPCUA/SimulationServer'. The message encoding/decoding limits imposed by the stack have been exceeded.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Information: Closing channel with endpoint 'opc.tcp://192.168.0.186:53530/OPCUA/SimulationServer'.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error closing channel. Object reference not set to an instance of an object.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Trace: Connecting in 20000 ms.
Thread finished: <Thread Pool> #4InspectorDebugSession(4): HandleTargetEvent: ThreadStopped

Savoyyang, 2017-04-25:
above is emulation debug

# **The following is use real phone debug**

[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Information: Discovering endpoints of 'opc.tcp://192.168.0.186:53530/OPCUA/SimulationServer'.

[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error discovering endpoints of 'opc.tcp://192.168.0.186:53530/OPCUA/SimulationServer'.The message encoding/decoding limits imposed by the stack have been exceeded.

[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Information: Closing channel with endpoint 'opc.tcp://192.168.0.186:53530/OPCUA/SimulationServer'.

[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error closing channel. Object reference not set to an instance of an object.

[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Trace: Connecting in 8000 ms.

04-25 09:22:34.067 D/AudioTrack(23561): write userSize 16384

04-25 09:22:34.160 D/AudioTrack(23561): write userSize 16384

04-25 09:22:34.253 D/AudioTrack(23561): write userSize 16384

04-25 09:22:34.345 D/AudioTrack(23561): write userSize 16384

04-25 09:22:34.438 D/AudioTrack(23561): write userSize 16384

04-25 09:22:34.532 D/AudioTrack(23561): write userSize 16384

04-25 09:22:34.628 D/AudioTrack(23561): write userSize 16384

04-25 09:22:34.720 D/AudioTrack(23561): write userSize 16384

04-25 09:22:34.813 D/AudioTrack(23561): write userSize 16384

04-25 09:22:34.906 D/AudioTrack(23561): write userSize 16384

04-25 09:22:34.999 D/AudioTrack(23561): write userSize 16384

04-25 09:22:35.098 D/AudioTrack(23561): write userSize 16384

04-25 09:22:35.183 D/AudioTrack(23561): write userSize 16384

04-25 09:22:35.279 D/AudioTrack(23561): write userSize 16384

04-25 09:22:35.370 D/AudioTrack(23561): write userSize 16384

04-25 09:22:35.467 D/AudioTrack(23561): write userSize 16384

Savoyyang, 2017-04-25:
# **Above is emulation debug
The following is real phone debug** 

[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error discovering endpoints of 'opc.tcp://192.168.3.132:53530/OPCUA/SimulationServer'. The message encoding/decoding limits imposed by the stack have been exceeded.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Information: Closing channel with endpoint 'opc.tcp://192.168.3.132:53530/OPCUA/SimulationServer'.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error closing channel. Object reference not set to an instance of an object.
04-25 09:52:32.458 D/AudioTrack(32485): write userSize 16384
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Trace: Connecting in 8000 ms.
04-25 09:52:32.555 D/AudioTrack(32485): write userSize 16384
04-25 09:52:32.646 D/AudioTrack(32485): write userSize 16384
04-25 09:52:32.736 D/AudioTrack(32485): write userSize 16384
04-25 09:52:32.829 D/AudioTrack(32485): write userSize 16384
04-25 09:52:32.922 D/AudioTrack(32485): write userSize 16384
04-25 09:52:33.014 D/AudioTrack(32485): write userSize 16384
04-25 09:52:33.107 D/AudioTrack(32485): write userSize 16384
04-25 09:52:33.203 D/AudioTrack(32485): write userSize 16384
04-25 09:52:33.293 D/AudioTrack(32485): write userSize 16384
04-25 09:52:33.386 D/AudioTrack(32485): write userSize 16384
04-25 09:52:33.479 D/AudioTrack(32485): write userSize 16384

awcullen, 2017-04-25:
I see the error is "The message encoding/decoding limits imposed by the stack have been exceeded".
That error was fixed in 1.5.12.
Check again that the package was upgraded to 1.5.12.

Savoyyang, 2017-04-25:
# I am sure the package is 1.5.12.  

[image]

# And I use Prosys Client connect the server should choose this option.

[image]

[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Information: Discovering endpoints of 'opc.tcp://192.168.1.103:53530'.
04-25 21:13:34.159 D/AudioTrack(29915): write userSize 16384
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error discovering endpoints of 'opc.tcp://192.168.1.103:53530'. The message encoding/decoding limits imposed by the stack have been exceeded.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Information: Closing channel with endpoint 'opc.tcp://192.168.1.103:53530'.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error closing channel. Object reference not set to an instance of an object.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Trace: Connecting in 8000 ms.
04-25 21:13:34.252 D/AudioTrack(29915): write userSize 16384
04-25 21:13:34.352 D/AudioTrack(29915): write userSize 16384
04-25 21:13:34.440 D/AudioTrack(29915): write userSize 16384
04-25 21:13:34.538 D/AudioTrack(29915): write userSize 16384
04-25 21:13:34.624 D/AudioTrack(29915): write userSize 16384

awcullen, 2017-04-25:
Just now, I downloaded a copy of opc-ua-samples project from the web site. I deleted old packages from the cache. I opened the StatusHmi project.  I restored the Nuget packages for the solution. I changed the URL in the app.config file to be "opc.tcp://bculz-pc:53530/OPCUA/SimulationServer". I ran it and it connected successfully.
```
Workstation.ServiceModel.Ua.UaTcpSessionClient: Information: Discovering endpoints of 'opc.tcp://bculz-pc:53530/OPCUA/SimulationServer'.
Workstation.ServiceModel.Ua.UaTcpSessionClient: Trace: Success discovering endpoints of 'opc.tcp://bculz-pc:53530/OPCUA/SimulationServer'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel opening.
Workstation.ServiceModel.Ua.DirectoryStore: Trace: Found certificate with subject alt name 'urn:ANDREW-THINK:Workstation.StatusHmi'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: Opening session channel with endpoint 'opc.tcp://bculz-PC:53530/OPCUA/SimulationServer'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityPolicy: 'http://opcfoundation.org/UA/SecurityPolicy#Basic256'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityMode: 'SignAndEncrypt'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: UserIdentityToken: 'Workstation.ServiceModel.Ua.AnonymousIdentity'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending OpenSecureChannelRequest Handle: 2
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received OpenSecureChannelResponse Handle: 2 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending CreateSessionRequest Handle: 3
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Installed new security token 1.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received CreateSessionResponse Handle: 3 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending ActivateSessionRequest Handle: 4
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received ActivateSessionResponse Handle: 4 Result: 0x00000000
```

Savoyyang, 2017-04-25:
I  also success in StatusHmi project,
you can text the Android APP.

awcullen, 2017-04-25:
Changed url to use ip:
```
        private string discoveryUrl = @"opc.tcp://192.168.1.109:53530/OPCUA/SimulationServer";
```
Success
```
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Information: Discovering endpoints of 'opc.tcp://192.168.1.109:53530/OPCUA/SimulationServer'.
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Trace: Success discovering endpoints of 'opc.tcp://192.168.1.109:53530/OPCUA/SimulationServer'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel opening.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: Opening session channel with endpoint 'opc.tcp://192.168.1.109:53530/OPCUA/SimulationServer'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityPolicy: 'http://opcfoundation.org/UA/SecurityPolicy#None'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityMode: 'None'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: UserIdentityToken: 'Workstation.ServiceModel.Ua.AnonymousIdentity'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending OpenSecureChannelRequest Handle: 2
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received OpenSecureChannelResponse Handle: 2 Result: 0x00000000
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending CreateSessionRequest Handle: 3
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Installed new security token 1.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received CreateSessionResponse Handle: 3 Result: 0x00000000
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending ActivateSessionRequest Handle: 4
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received ActivateSessionResponse Handle: 4 Result: 0x00000000
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending ReadRequest Handle: 5
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received ReadResponse Handle: 5 Result: 0x00000000
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel opened.
```

Savoyyang, 2017-04-25:
Just now uninstall Workstation.UaClient and restore to 1.5.12, than I trust it,

[image]

But I try it again ,it still display the error

[image]

Savoyyang, 2017-04-25:
Just now I try many times(use IP),but error still exit. 
I use the full name but it return ip+port.

[image]

awcullen, 2017-04-25:
Try deleting the \bin and \obj directories of the android project. 
Because the change to discoveryUrl is not showing in the debug output, you can be sure that the new project is not being deployed to the device.

Savoyyang, 2017-04-26:
Yes, I deleting the \bin and \obj directories of the android project, and it works.
Another good news that Android APP can connect Beckhoff Twincat OpcUa Server.I tested.

[0:] Workstation.ServiceModel.Ua.DirectoryStore: Trace: Found certificate with subject alt name 'urn:localhost:Workstation.MobileHmi'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: Opening session channel with endpoint 'opc.tcp://192.168.3.117:53530/OPCUA/SimulationServer'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityPolicy: 'http://opcfoundation.org/UA/SecurityPolicy#Basic256'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityMode: 'SignAndEncrypt'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: UserIdentityToken: 'Workstation.ServiceModel.Ua.AnonymousIdentity'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending OpenSecureChannelRequest Handle: 2

awcullen, 2017-04-26:
Good news indeed. Thanks for not giving up!
Closing issue.

closed 2017-04-26 by awcullen