convertersystems/opc-ua-samples #24: .NET Standard - based sample

abrasat, 2017-04-25:
Are you planning also to add a sample with the usage of UaClient in a .NET Standard based project ?
For instance Xamarin.Forms can use .NET Standard 1.4 based libraries, as described [here](https://oren.codes/2017/04/23/using-xamarin-forms-with-net-standard-vs-2017-edition/ )

awcullen, 2017-04-25:
Are we ready to upgrade to VS 2017 only projects in this repo?

abrasat, 2017-04-25:
I am working with VS2017 official release and it seems ok. I did not experienced the problems that I had with VS 2017 RC.

awcullen, 2017-04-26:
I think we should leave the Xamarin.Forms sample 'MobileHmi' as a shared project. An important part of the sample is the robot, which requires UrhoSharp. UrhoSharp does not support the NETStandard platform. It requires native libraries for Android, iOS, Win, UWP.

Closing this issue for now.

closed 2017-04-26 by awcullen

Savoyyang, 2017-05-03:
If don't requires UrhoSharp , can I change shared project to .NET Standard /PCL project？

awcullen, 2017-05-03:
You can use UaClient in a .NET Standard project (target framework 1.4 or higher). I don't think you can use UaClient in a PCL project.