convertersystems/opc-ua-samples #25: Fetch data to mobile from Cloud

tejasklodha, 2017-04-27:
Dear Sir,

Its not issue but want suggestion.
I want to use mobile application MobileHmi from any where through internet.

because MobileHmi.Droid only run in intranet(in the same network) but how we can use this application in internet?

Any OPC UA web service required? 

Thanks.

awcullen, 2017-04-27:
Now, you could open your opc ua server to the internet (with security turned on, I hope). The current protocol was designed for speed (binary) and security (encryption) when communicating with servers over the internet.

I am anticipating the release of opc ua servers that support the PubSub api. With this new extension to the protocol, you can use a cloud service to be the intermediary between a client and server. 

As a solution for today, I have been hearing good things about https://thingsboard.io/

tejasklodha, 2017-04-29:
Thanks awcullen for giving information.. I will look that....
Also start working in azure with OPC....

closed 2017-04-29 by tejasklodha

tejasklodha, 2017-05-09:
Dear Sir,

I found some interesting information about OPC UA on internet.

https://github.com/node-opcua/node-opcua/wiki/publicly-available-OPCUA-servers