convertersystems/opc-ua-samples #26: Can possible that many instants of OPC UA server fetch and display in mobile app ?

tejasklodha, 2017-04-29:
Dear Sir,

I am looking for that **multiple opc ua server** fetch and display **on same page of mobile application**

Example: 
I have 10 OPC UA server and fetch some 5 variables from each server and display 5 variable of each opc ua server on same page of android application. can you demonstrate this in Workstation.MobileHmi ? 

Thanks,

awcullen, 2017-04-29:
When using version 2.0.0-RC2, you can build a UaApplication in app.cs.
```
        protected override void OnStart()
        {
            // Setup a logger.
            this.loggerFactory = new LoggerFactory();
            this.loggerFactory.AddDebug(LogLevel.Trace);

            // Build and run an OPC UA application instance.
            this.application = new UaApplicationBuilder()
                .UseApplicationUri($"urn:{Dns.GetHostName()}:Workstation.MobileHmi")
                .UseDirectoryStore(Path.Combine(
                    Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData),
                    "pki"))
                .UseIdentity(this.ShowSignInDialog)
                .UseLoggerFactory(this.loggerFactory)
                .Build();

            this.application.Run();

            // Show the MainPage
            this.MainPage = new NavigationPage(new MainPage());
        }

```

Then you create 10 view-models, each with a specific name and Subscription attribute with the ipAddress of the server.
```
    /// <summary>
    /// A model for PLC1MainView.
    /// </summary>
    [Subscription(endpointUrl: "opc.tcp://plc1:26543", publishingInterval: 500, keepAliveCount: 20)]
    public class PLC1MainViewModel : SubscriptionBase
    {
        /// <summary>
        /// Gets or sets the value of Axis1.
        /// </summary>
        [MonitoredItem(nodeId: "ns=2;s=Axis1")]
        public float Axis1
        {
            get { return this.axis1; }
            set { this.SetProperty(ref this.axis1, value); }
        }
        private float axis1;
}
```

The last thing is the MainView, which shows the 10 servers
```
        <ScrollView>
          <StackLayout">
            <StackLayout">
                <StackLayout.BindingContext>
                    <local:PLC1MainViewModel />
                </StackLayout.BindingContext>
                <Label Text="PLC1Axis1" />
                <Slider Minimum="-180"
                        Maximum="180"
                        Value="{Binding Axis1, Mode=TwoWay}"/>
            </StackLayout">

            <StackLayout">
                <StackLayout.BindingContext>
                    <local:PLC2MainViewModel />
                </StackLayout.BindingContext>
                <Label Text="PLC2Axis1" />
                <Slider Minimum="-180"
                        Maximum="180"
                        Value="{Binding Axis1, Mode=TwoWay}"/>
            </StackLayout">

            <!--repeat 8 more times-->

          </StackLayout">
        </ScrollView>

```

tejasklodha, 2017-05-02:
Dear Sir,

Above method work and it is very easy compare to previous code ... but sometimes it is worked and some time not working..

Means I made android and UWP mobile app which display two OPC UA server data at same time and when i am trying to run UWP app its working fine and display both PLC data simultaneously real time but same application I run again without change anything in code but anyone server not update value means slider not change even i can able to browse OPC UA server from OPC Browser. same issue observed in android mobile app

I also observe that when OPC UA server not change dynamic value from OPC UA sever at that time NO Session created.

Can you solve this issue ?

Thanks,

awcullen, 2017-05-02:
There should be lots of information in the Debug output window. Please save the output to a file and post here or on Dropbox.

tejasklodha, 2017-05-02:
Dear sir,

Find below output from VS 15.
Note:  I am able to Browse my OPC UA Server by using OPC Browser  

[Android Output.txt](https://github.com/convertersystems/opc-ua-samples/files/970542/Android.Output.txt)
[Android Output 2.txt](https://github.com/convertersystems/opc-ua-samples/files/970543/Android.Output.2.txt)
[UWP Output1.txt](https://github.com/convertersystems/opc-ua-samples/files/970544/UWP.Output1.txt)

awcullen, 2017-05-02:
For UWP:
UaTcpSecureChannel:Trace: Created certificate with subject alt name 'urn:DESKTOP-L7MTVKN:myIoTUWP'.
UaTcpSecureChannel:Trace: Sending OpenSecureChannelRequest, Handle: 2
UaTcpSecureChannel:Error: Error receiving response. [BadSecurityChecksFailed]

I know that error to mean that the UA Server did not accept the certificate.

For Droid:
There is no logging of the UaTcpSecureChannel.  Please verify you call this.application.Run() in the OnStart() method.

tejasklodha, 2017-05-03:
Dear Sir,

**For android.**

I already mentioned this.application.Run() in the OnStart() method but still sometime code works and some time code not works.
at same time both OPC UA server I can able to browse in OPC UA browser.

below are my code.
App.cs call PLC1View where  I bind two view model (MainPageViewModel, PLC1MainViewModel ) in stacklayout 

[PLC1MainViewModel.cs.txt](https://github.com/convertersystems/opc-ua-samples/files/972474/PLC1MainViewModel.cs.txt)
[PLC1View.xaml.txt](https://github.com/convertersystems/opc-ua-samples/files/972475/PLC1View.xaml.txt)
[PLC1View.xaml.cs.txt](https://github.com/convertersystems/opc-ua-samples/files/972476/PLC1View.xaml.cs.txt)
[App.xaml.cs.txt](https://github.com/convertersystems/opc-ua-samples/files/972477/App.xaml.cs.txt)
[MainPageViewModel.cs.txt](https://github.com/convertersystems/opc-ua-samples/files/972478/MainPageViewModel.cs.txt)

tejasklodha, 2017-05-03:
Dear Sir,

For UWP

Below are today's debug logs and application screenshot for your reference.
when I run application first time then one PLC simulator data updated real time and I can able to view on screen as per attached screenshot and after that again when i run same application without change of code then both PLC data not updated in application as per attached screenshot  and again note that at same time both PLC data(OPC UA server data), I can able to browse in OPC UA browser. .
I attached both case debug output and application screenshot as below.

[UWP Output 03_05 Debug 2.txt](https://github.com/convertersystems/opc-ua-samples/files/972491/UWP.Output.03_05.Debug.2.txt)
[image]
[image]
[UWP Output 03_05 Debug 1.txt](https://github.com/convertersystems/opc-ua-samples/files/972492/UWP.Output.03_05.Debug.1.txt)

Thanks,

tejasklodha, 2017-05-03:
Dear Sir ,

For Android.

I add one main_page that display only one view model also also add one button which can redirect PLC1View page as mentioned above 
 below are the screenshot of main_page (Android Screenshoot main_page )

[image]

 below are the screenshot of PLCView (Android Screenshoot PLCView page  )

[image]

below are debug output of android application 

[Android Output 03_05 Debug 1.txt](https://github.com/convertersystems/opc-ua-samples/files/972558/Android.Output.03_05.Debug.1.txt)

Thanks,

tejasklodha, 2017-05-03:
Dear Sir,

For UWP

More Debug Logs for your reference...

Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Information: Discovering endpoints of 'opc.tcp://192.168.137.1:26543'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel closed.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel closed.
The thread 0x3910 has exited with code 0 (0x0).
The thread 0x3300 has exited with code 0 (0x0).
The thread 0xb4c has exited with code 0 (0x0).
The thread 0x2148 has exited with code 0 (0x0).
The thread 0x29a8 has exited with code 0 (0x0).
The thread 0x2484 has exited with code 0 (0x0).
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Error: Error discovering endpoints of 'opc.tcp://192.168.137.1:26543'. A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond 192.168.137.1:26543
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel faulted.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel closing.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel closed.
IoT_Mobile.MainPageViewModel:Trace: Error getting channel. A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond 192.168.137.1:26543
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Information: Discovering endpoints of 'opc.tcp://192.168.137.1:26543'.
The thread 0x12e8 has exited with code 0 (0x0).
The thread 0x1fd8 has exited with code 0 (0x0).
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Error: Error discovering endpoints of 'opc.tcp://192.168.137.1:26543'. A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond 192.168.137.1:26543
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel faulted.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel closing.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel closed.
IoT_Mobile.MainPageViewModel:Trace: Error getting channel. A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond 192.168.137.1:26543
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Information: Discovering endpoints of 'opc.tcp://192.168.137.1:26543'.
The thread 0x3420 has exited with code 0 (0x0).
The thread 0x26bc has exited with code 0 (0x0).
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Information: Discovering endpoints of 'opc.tcp://192.168.137.1:48030'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Error: Error discovering endpoints of 'opc.tcp://192.168.137.1:26543'. A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond 192.168.137.1:26543
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel faulted.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel closing.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel closed.
IoT_Mobile.MainPageViewModel:Trace: Error getting channel. A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond 192.168.137.1:26543
IoT_Mobile.MainPageViewModel:Trace: Error getting channel. A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond 192.168.137.1:26543
IoT_Mobile.MainPageViewModel:Trace: Error getting channel. A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond 192.168.137.1:26543
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Information: Discovering endpoints of 'opc.tcp://192.168.137.1:26543'.
The thread 0x21d8 has exited with code 0 (0x0).
The thread 0x26f0 has exited with code 0 (0x0).
The thread 0x2f20 has exited with code 0 (0x0).
The thread 0x3a94 has exited with code 0 (0x0).
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Error: Error discovering endpoints of 'opc.tcp://192.168.137.1:48030'. A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond 192.168.137.1:48030
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel faulted.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel closing.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Trace: Channel closed.
IoT_Mobile.PLC1MainViewModel:Trace: Error getting channel. A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond 192.168.137.1:48030
[... cut ...]
IoT_Mobile.MainPageViewModel:Trace: Error getting channel. A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond 192.168.137.1:26543
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel:Information: Discovering endpoints of 'opc.tcp://192.168.137.1:26543'.
The thread 0x1e34 has exited with code 0 (0x0).
The thread 0x37f8 has exited with code 0 (0x0).
The thread 0x353c has exited with code 0 (0x0).
The thread 0xb60 has exited with code 0 (0x0).
The thread 0x1230 has exited with code 0 (0x0).
The thread 0x1e88 has exited with code 0 (0x0).

tejasklodha, 2017-05-04:
**Dear Sir,**

**Today I again debug code and still found random behavior of code.**
**Sometime both PLC data display on scren and same code run again then some time both PLC can not show real time data**
**Below are Debug code for both situation**

## **Situation 1 : Debug code when Both PLC  data update Real time on screen of UWp app** 
 
'IoT_Mobile.UWP.exe' (CoreCLR: DefaultDomain): Loaded 'C:\Program Files\WindowsApps\Microsoft.NET.CoreRuntime.1.0_1.0.23819.0_x86__8wekyb3d8bbwe\mscorlib.ni.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\entrypoint\IoT_Mobile.UWP.exe'. Symbols loaded.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\System.Runtime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\WinMetadata\Windows.winmd'. Module was built without symbols.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\System.Runtime.InteropServices.WindowsRuntime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\System.Diagnostics.Debug.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\System.Runtime.WindowsRuntime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\System.Runtime.WindowsRuntime.UI.Xaml.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\Xamarin.Forms.Platform.UAP.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\System.Reflection.dll'. Module was built without symbols.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\Xamarin.Forms.Core.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\System.Threading.Tasks.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\System.IO.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\System.Private.Uri.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\System.ObjectModel.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\System.Collections.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\System.Linq.Expressions.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'IoT_Mobile.UWP.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Users\FMIL\Documents\Visual Studio 2015\Projects\IoTUWP\IoT_Mobile\IoT_Mobile.UWP\bin\x86\Debug\AppX\System.Linq.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
[... cut ...]
IoT_Mobile.PLC1MainViewModel:Trace: Error getting channel. A task was canceled.
IoT_Mobile.PLC1MainViewModel:Trace: Error getting channel. A task was canceled.
IoT_Mobile.PLC1MainViewModel:Trace: Error getting channel. A task was canceled.
The thread 0x9d0 has exited with code 0 (0x0).
IoT_Mobile.PLC1MainViewModel:Trace: Error getting channel. A task was canceled.
IoT_Mobile.PLC1MainViewModel:Trace: Error getting channel. A task was canceled.

**Any solution you found ?**

**Let me know if you want more debug logs**

Thanks,

awcullen, 2017-05-04:
Thank you for the debug logs. I found the bug and am working on fixing it.

awcullen, 2017-05-06:
Try version 2.0.0-RC4 please.

tejasklodha, 2017-05-07:
Dear Sir,

**Its working Fine in UWP mobile app but not expected working in Android mobile app.**
I also tried to run after deleting bin and object folder content

**find below debug logs of android real device for your reference.**

**On display I have Display two OPC server but only one opc server data change in real time** 

Android application is debugging.
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/IoT_Mobile.Android.dll
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/BouncyCastle.Crypto.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/FormsViewGroup.dll [External]
05-07 16:43:35.721 D/ActivityThread( 5224): installProvider: context.getPackageName()=IoT_Mobile.Android
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Microsoft.Extensions.Configuration.Abstractions.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Microsoft.Extensions.Configuration.Binder.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Microsoft.Extensions.Configuration.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Microsoft.Extensions.DependencyInjection.Abstractions.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Microsoft.Extensions.Logging.Abstractions.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Microsoft.Extensions.Logging.Debug.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Microsoft.Extensions.Logging.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Microsoft.Extensions.Primitives.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/SkiaSharp.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/SkiaSharp.Views.Android.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/SkiaSharp.Views.Forms.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/System.Reactive.Core.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/System.Reactive.Interfaces.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/System.Reactive.Linq.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/System.Reactive.PlatformServices.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/System.Runtime.CompilerServices.Unsafe.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/System.Threading.Tasks.Dataflow.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Telerik.Xamarin.Android.Chart.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Telerik.Xamarin.Android.Common.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Telerik.Xamarin.Android.Data.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Telerik.Xamarin.Android.Input.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Telerik.Xamarin.Android.List.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Telerik.Xamarin.Android.Primitives.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Telerik.XamarinForms.Chart.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Telerik.XamarinForms.Common.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Telerik.XamarinForms.DataControls.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Telerik.XamarinForms.DataVisualization.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Telerik.XamarinForms.Input.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Telerik.XamarinForms.Primitives.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Telerik.XamarinForms.SkiaSharp.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Urho.Droid.SdlBinding.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Urho.Forms.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Workstation.UaClient.dll [External]
Loaded assembly: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/Xamarin.Android.Support.Animated.Vector.Drawable.dll [External]
[... cut ...]
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending PublishRequest, Handle: 60
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received WriteResponse, Handle: 56 Result: 0x00000000
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received WriteResponse, Handle: 57 Result: 0x00000000
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received WriteResponse, Handle: 58 Result: 0x00000000
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending WriteRequest, Handle: 61

Thanks,

awcullen, 2017-05-07:
Your UWP app can connect to 'opc.tcp://192.168.0.6:48030', but your Android app can not?
Then '48030' is probably not accepting the Android app's certificate.
What is the vender of server at 48030?

tejasklodha, 2017-05-08:
Dear Sir,
### Its working like charm... Thanks....
Now its working in android as well UWP without any issue.

Below steps I follow.

clean solution
delete bin and obj and rebuild
downgrade and upgrade package of client again.

**Then issue resolved.**

Vendor of OPC UA server is https://www.unified-automation.com/

closed 2017-05-08 by tejasklodha