convertersystems/opc-ua-samples #27: UWP app rebuild issue in Visual Studio 2015

tejasklodha, 2017-05-02:
Dear Sir,

When I am download latest sample opc-ua-samples on today and try to rebuild in my visual studio then give below error 

_Error The Microsoft.Net.Native.Compiler package, included in Microsoft.NETCore.UniversalWindowsPlatform 5.3 and later, is only supported in Visual Studio 2017 and later. You can downgrade the version of the Microsoft.NETCore.UniversalWindowsPlatorm NuGet package to 5.2.* or earlier for Visual Studio 2015.	Workstation.Common.Uwp_	

I downgrade **Microsoft.NETCore.UniversalWindowsPlatform** to **5.1.0 , 5.2.2 , 5.2.3** but still getting same error...

also delete the all content from bin and obj folder and try to rebuild but getting same errors.
What should I do if i want to run code in Visual studio 2015?

Thanks,

awcullen, 2017-05-02:
Check if your building a Debug or Release build.
If you wish to compile a Release Build you can open UWP project properties and on the Build tab you can uncheck the [ ] Compile with .NET Native Tool Chain

tejasklodha, 2017-05-09:
Dear Sir,

Resolved issue after update 2.0.0-RC5

Thanks,

closed 2017-05-09 by tejasklodha