convertersystems/opc-ua-samples #28: Can we send live data dump to Sql Server from PLC

tejasklodha, 2017-05-04:
Dear Sir,

It is not issue but enhancement.

Can you demonstrate that live variable of data from PLC to send sql server and store in database?

Thanks,

tejasklodha, 2017-05-16:
Dear Sir,

Its really very helpful to us if you Can demonstrate this..

Thanks,

awcullen, 2017-05-16:
I do plan to add this feature to the DataLogger sample.

In the meantime, I am fascinated with the open source project [https://thingsboard.io/](https://thingsboard.io/).

I think you have tried Azure IOT, correct?

Well, Thingsboard is similar, in that it collects IOT data, stores in database, has server functions, and dashboard-style web hmi. In addition, it can run in cloud -or- completely within your office / factory. 
In my office, I have a Kepware OPC UA server providing simulated data. A service 'tb-gateway', creates subscription and forwards the data to 'thingsboard' service via MQTT. 'Thingsboard' service stores in a database and updates any web connected dashboard hmi's. Try for yourself.

tejasklodha, 2017-05-17:
Dear sir,

Sounds good to listen that you are planning to add this feature :).... 

Yup I already tried azure iot but first main requirement is store opc ua data in local database only(no cloud).... Just add your comments when you completed...

and right now i take reference of your code and start working to make new wpf application which store opc ua server data in local sql server database​ but I will also explore to thingsboard.io.......

Thanks.

tejasklodha, 2017-07-18:
Done...

closed 2017-07-18 by tejasklodha