convertersystems/opc-ua-samples #30: Fetch Live data from method

tejasklodha, 2017-05-08:
Dear Sir,

I can able to fetch live data via viewmodel.
But how can I fetch live data through method which call to Viewmodel from diffrent class

Example

I have one main view model **MainPageViewModel** 

```

using System;
using System.Runtime.CompilerServices;
using Workstation.ServiceModel.Ua;

namespace IoT_Mobile
{

    [Subscription(endpointUrl: "opc.tcp://192.168.137.1:26543", publishingInterval: 501, keepAliveCount: 30)]

    public class **MainPageViewModel** : SubscriptionBase
    {

        [MonitoredItem(nodeId: "ns=2;s=Robot1_Axis4")]
        public float Robot1Axis4
        {
            get { return this.robot1Axis4; }
            set { this.SetPropertyWithDeadband(ref this.robot1Axis4, value, 1.0f); }
        }

        private float robot1Axis4;

        /// <summary>
        /// Added to help with Xamarin Forms implementation of Slider.
        /// </summary>
        protected virtual bool SetPropertyWithDeadband(ref float storage, float value, float deadband, [CallerMemberName] string propertyName = null)
        {
            if (Math.Abs(storage - value) < deadband)
            {
                return false;
            }

            storage = value;
            this.NotifyPropertyChanged(propertyName);
            return true;
        }
    }
}
```

**Now I want Live data through method GetData() as below from different class**

```
 public static double GetValue()
        {
            MainPageViewModel obj1 = new MainPageViewModel();
            double axiesdata = obj1.Robot1Axis1;

            return axiesdata;
        }

```
But every time I got value 0 from above method.
Any mistake i Did in above?

need your help.

thanks,

awcullen, 2017-05-08:
The short answer is that the ViewModel is not initially connected to the server when it is constructed. Connecting and subscribing happens asynchronously on a threadpool thread. If you wish to read the value of Axis1 in a method, then you could
```
        private static async Task<double> GetValueAsync()
        {
            var channel = await UaApplication.Current.GetChannelAsync("opc.tcp://localhost:26543");
            var result = await channel.ReadAsync(new ReadRequest
            {
                NodesToRead = new[] {
                    new ReadValueId { NodeId = NodeId.Parse("ns=2;s=Robot1_Axis1"), AttributeId = AttributeIds.Value }
                }
            });
            double axisdata = result.Results[0].GetValueOrDefault<float>();
            return axisdata;
        }

```

The long answer is that initially SubscriptionBase is monitoring for when a view adds a eventhandler to PropertyChanged event. When a view adds an eventhandler, then SubscriptionBase creates the subscription. When the view removes the eventhandler, then SubscriptionBase deletes the subscription.
So if you are creating a viewmodel, but not using a view (with bindings to the viewmodel), then you must add an eventhandler yourself if you want the SubscriptionBase to work.
 
```
            var vm= new MainPageViewModel();
            var d = new PropertyChangedEventHandler((s, e) => { });
            // create the subscription.
            vm.PropertyChanged += d;

            // and when you want the subscription deleted.
           sub.PropertyChanged -= d;

```

tejasklodha, 2017-05-08:
Dear Sir,

thanks for detail explanation.
I tried below code

```
 public static async Task<double> GetValueAsync()
        {
            var channel = await UaApplication.Current.GetChannelAsync("opc.tcp://192.168.0.6:26543");
            var result = await channel.ReadAsync(new ReadRequest
            {
                
                NodesToRead = new[] {
                    new ReadValueId { NodeId = NodeId.Parse("ns=2;s=Robot1_Axis1"), AttributeId = AttributeIds.Value }
                }
            });
            double axisdata = result.Results[0].GetValueOrDefault<float>();
            return axisdata;
        }

```
and then I want to get value in real time  so I create variable and tried to fetch value by using below code

` double result = GetValueAsync().Result;`

but still unable to succeed to get value. Am I doing anything wrong?

Thanks,

awcullen, 2017-05-09:
You are not doing anything wrong. The library had a bug in one method that caused a deadlock with your code. Please update to version 2.0.0-RC5.

And thanks for reporting it!

tejasklodha, 2017-05-09:
Dear Sir,

Update package  2.0.0-RC5 and Tried same again but still not able to fetch data.

Thanks,

tejasklodha, 2017-05-09:
**Dear Sir,**

**Find below debug log for detail troubleshooting.**

```
Android application is debugging.
05-09 09:22:28.587 I/art     (21278): Late-enabling -Xcheck:jni
05-09 09:22:28.617 D/ActivityThread(21278): installProvider: context.getPackageName()=IoT_Mobile.Android
05-09 09:22:28.652 W/monodroid(21278): Creating public update directory: `/data/user/0/IoT_Mobile.Android/files/.__override__`
05-09 09:22:28.652 W/monodroid(21278): Using override path: /data/user/0/IoT_Mobile.Android/files/.__override__
05-09 09:22:28.653 W/monodroid(21278): Using override path: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__
05-09 09:22:28.653 W/monodroid(21278): Trying to load sgen from: /data/user/0/IoT_Mobile.Android/files/.__override__/libmonosgen-2.0.so
05-09 09:22:28.653 W/monodroid(21278): Trying to load sgen from: /storage/emulated/0/Android/data/IoT_Mobile.Android/files/.__override__/libmonosgen-2.0.so
05-09 09:22:28.653 W/monodroid(21278): Trying to load sgen from: /data/app/IoT_Mobile.Android-1/lib/arm/libmonosgen-2.0.so
05-09 09:22:28.653 W/monodroid(21278): Trying to load sgen from: /data/user/0/IoT_Mobile.Android/files/.__override__/links/libmonosgen-2.0.so
05-09 09:22:28.658 W/monodroid-debug(21278): Trying to initialize the debugger with options: --debugger-agent=transport=dt_socket,loglevel=0,address=127.0.0.1:8937,server=y,embedding=1
05-09 09:22:28.812 W/monodroid-debug(21278): Accepted stdout connection: 22
05-09 09:22:28.848 D/libc-netbsd(21278): [getaddrinfo]: mtk hostname=127.0.0.1; servname=8937; netid=0; mark=0
05-09 09:22:28.848 D/libc-netbsd(21278): getaddrinfo( app_uid:10457
05-09 09:22:28.848 D/libc-netbsd(21278): getaddrinfo() uid prop:
05-09 09:22:28.848 D/libc-netbsd(21278): getaddrinfo() getuid():10457
05-09 09:22:28.848 D/libc-netbsd(21278): [getaddrinfo]: mtk ai_addrlen=0; ai_canonname=(null); ai_flags=0; ai_family=0
.
.
.
.
.
.
.Load Assembly
.
.
.
.
.
05-09 09:22:34.892 D/Mono    (21278): Searching for 'java_interop_jnienv_get_static_object_field'.
05-09 09:22:34.892 D/Mono    (21278): Probing 'java_interop_jnienv_get_static_object_field'.
05-09 09:22:34.892 D/Mono    (21278): Found as 'java_interop_jnienv_get_static_object_field'.
05-09 09:22:35.037 D/Mono    (21278): Assembly Ref addref Telerik.XamarinForms.ChartRenderer.Android[0xe8143060] -> System.Core[0xd3956520]: 11
05-09 09:22:35.064 D/Mono    (21278): DllImport searching in: '__Internal' ('(null)').
05-09 09:22:35.064 D/Mono    (21278): Searching for 'java_interop_jnienv_call_static_object_method'.
05-09 09:22:35.064 D/Mono    (21278): Probing 'java_interop_jnienv_call_static_object_method'.
05-09 09:22:35.064 D/Mono    (21278): Found as 'java_interop_jnienv_call_static_object_method'.
05-09 09:22:35.078 D/Mono    (21278): DllImport searching in: '__Internal' ('(null)').
05-09 09:22:35.078 D/Mono    (21278): Searching for 'java_interop_jnienv_get_object_array_element'.
05-09 09:22:35.078 D/Mono    (21278): Probing 'java_interop_jnienv_get_object_array_element'.
05-09 09:22:35.078 D/Mono    (21278): Found as 'java_interop_jnienv_get_object_array_element'.
05-09 09:22:35.149 D/Mono    (21278): DllImport searching in: '__Internal' ('(null)').
05-09 09:22:35.149 D/Mono    (21278): Searching for 'java_interop_jnienv_new_local_ref'.
05-09 09:22:35.149 D/Mono    (21278): Probing 'java_interop_jnienv_new_local_ref'.
05-09 09:22:35.149 D/Mono    (21278): Found as 'java_interop_jnienv_new_local_ref'.
05-09 09:22:35.222 D/Mono    (21278): Unloading image System.Threading.dll [0xd1c1ad00].
05-09 09:22:35.222 D/Mono    (21278): Image addref System.Threading[0xd1c3b460] -> System.Threading.dll[0xd4a46b00]: 8
05-09 09:22:35.222 D/Mono    (21278): Config attempting to parse: 'System.Threading.dll.config'.
05-09 09:22:35.222 D/Mono    (21278): Config attempting to parse: '/Users/builder/data/lanes/4468/f913a78a/source/monodroid/builds/install/mono-armv7/etc/mono/assemblies/System.Threading/System.Threading.config'.
05-09 09:22:35.222 D/Mono    (21278): Assembly Ref addref Telerik.XamarinForms.Chart[0xe8143000] -> System.Threading[0xd3c63420]: 8
05-09 09:22:35.261 D/Mono    (21278): DllImport searching in: '__Internal' ('(null)').
05-09 09:22:35.261 D/Mono    (21278): Searching for 'java_interop_jnienv_call_nonvirtual_boolean_method'.
05-09 09:22:35.261 D/Mono    (21278): Probing 'java_interop_jnienv_call_nonvirtual_boolean_method'.
05-09 09:22:35.261 D/Mono    (21278): Found as 'java_interop_jnienv_call_nonvirtual_boolean_method'.
05-09 09:22:35.268 D/Mono    (21278): DllImport searching in: '__Internal' ('(null)').
05-09 09:22:35.268 D/Mono    (21278): Searching for 'java_interop_jnienv_call_nonvirtual_void_method'.
05-09 09:22:35.268 D/Mono    (21278): Probing 'java_interop_jnienv_call_nonvirtual_void_method'.
05-09 09:22:35.268 D/Mono    (21278): Found as 'java_interop_jnienv_call_nonvirtual_void_method'.
05-09 09:22:35.270 D/ActivityThread(21278): ACT-AM_ON_RESUME_CALLED ActivityRecord{bbcca43 token=android.os.BinderProxy@7d4b8c0 {IoT_Mobile.Android/md5f00a4c94c09674c4d77a41c06c316ff4.MainActivity}}
05-09 09:22:35.270 V/ActivityThread(21278): Resume ActivityRecord{bbcca43 token=android.os.BinderProxy@7d4b8c0 {IoT_Mobile.Android/md5f00a4c94c09674c4d77a41c06c316ff4.MainActivity}} started activity: false, hideForNow: false, finished: false
[... cut ...]
Resolved pending breakpoint at 'BarSeriesPage.xaml.cs:126,1' to void IoT_Mobile.BarSeriesPage.<GetValueAsync>d__6.MoveNext () [0x001b2].
[0:] Workstation.ServiceModel.Ua.UaApplication: Trace: Begin getting UaTcpSessionChannel for opc.tcp://192.168.137.1:26543**
05-09 09:23:16.153 D/Mono    (21278): [0xd223f930] worker finishing
Thread finished: <Thread Pool> #4
The thread 'Unknown' (0x4) has exited with code 0 (0x0).
Thread finished: <Thread Pool> #2
The thread 'Unknown' (0x2) has exited with code 0 (0x0).

```

Thanks,

awcullen, 2017-05-09:
The trace shows:
```
[0:] Workstation.ServiceModel.Ua.UaApplication: Trace: UaApplication suspended.
[0:] Workstation.ServiceModel.Ua.UaApplication: Trace: Begin getting UaTcpSessionChannel for opc.tcp://192.168.137.1:26543**
```
There's a lot missing. Also, why does the url end with '**'?

Be sure the application has the following at least.

```
    public class App : Application
    {
        private ILoggerFactory loggerFactory;
        private UaApplication application;

        protected override void OnStart()
        {
            // Setup a logger.
            this.loggerFactory = new LoggerFactory();
            this.loggerFactory.AddDebug(LogLevel.Trace);

            // Build and run an OPC UA application instance.
            this.application = new UaApplicationBuilder()
                .UseApplicationUri($"urn:{Dns.GetHostName()}:Workstation.MobileHmi")
                .UseDirectoryStore(Path.Combine(
                    Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData),
                    "pki"))
                .UseLoggerFactory(this.loggerFactory)
                .Build();

            this.application.Run();

            // Show the MainPage
            this.MainPage = new NavigationPage(new MainPage());
        }

        protected override void OnSleep()
        {
            this.application?.SuspendAsync().Wait();
        }

        protected override void OnResume()
        {
            this.application?.Run();
        }
    }
```

tejasklodha, 2017-05-09:
** is just added when i use bold in github.... In code thare is no **
included

And I already mentioned and I called first mainpage with mainview model and
then put one button which is redirect the page when I call custom method...

Note: mainpage display real time varible....

Thanks,

awcullen, 2017-05-09:
Would you kindly share your project? 

You can compress (zip) the solution folder and attach (drag and drop) the file to your response.

tejasklodha, 2017-05-09:
Dear Sir,

Kindly Find Full Source code in attachment

below source code is without package folder
[IoTUWP.zip](https://github.com/convertersystems/opc-ua-samples/files/987773/IoTUWP.zip)

below source code is with package folder.
https://drive.google.com/file/d/0Bx0zJbhIo69vclc1MHREWkozOWs/view?usp=sharing

Let me know if you have any issue in Source code

Thanks,

awcullen, 2017-05-09:
I have fixed up your source code. Everything works well now.
[IoTUWP.zip](https://github.com/convertersystems/opc-ua-samples/files/988309/IoTUWP.zip)
[image]
[image]

tejasklodha, 2017-05-10:
Dear @awcullen ,

Working Fine.
I truly appreciate your time and efforts.

Thanks.

closed 2017-05-10 by tejasklodha

tejasklodha, 2017-05-11:
Dear Sir,

If I want to get status then I use below code.
when Status is Opened then I put below method under onButtonclik then its working fine and show in alert box with live status is Opened
But Problem is When another status the Opened then application crash. 

Below is code for your reference

```
 public partial class MainPage : ContentPage
    {
        static CommunicationState state;
        public MainPage()
        {

            InitializeComponent();
        }

        public void OnClikedStatus(object sender, EventArgs e)
        {
            DisplayAlert("Status : ", "Current status is " + Getmachinestatus().Result, "Ok");
        }

        public static async Task<CommunicationState> Getmachinestatus()
        {
            // added .ConfigureAwait(false) to the awaited methods to avoid deadlock with the GetValueAsync.Result;
            var channel = await UaApplication.Current.GetChannelAsync("opc.tcp://192.168.137.1:26543").ConfigureAwait(false);
            state = channel.State;

            return state;
        }
        

    }
```

Also I want to show **errors/validation**  on display like
Server not found,  
Node ID not found , 
Endpoint required user and password and show dialog box for authentication.

How We can achieve this?

 Thanks,

awcullen, 2017-05-11:
In your method Getmachinestatus(), calling GetChannelAsync() will not return until the state is Open. So, its not a good way to check if the state is not open.

I suggest you use SubscriptionBase, like you have done with MainPageViewModel. With MainPageViewModel you can bind to property State, and you can use INotifyDataErrorInfo to identify NodeIds that are not found.
I plan to add a dialog for UserName/Password to the MobileHmi app. It will be similar to the other samples.

I can add a NavigationPage

tejasklodha, 2017-05-11:
For Getmachinestatus

I can able to getdata with viewmodel and its working fine but i just want this live info on View(not view model)  because i want to use this method in many others methods....

For Errors
Can u give examples that how use INotifyDataErrorInfo and get errors

For Authentication/Dialogue
Sound good :)

Thank you so much... :)

awcullen, 2017-05-11:
SubscriptionBase implements System.ComponentModel.INotifyDataErrorInfo.
When there is a error subscribing to a Node or writing to a node, I put a message in the ErrorsCollection.

WPF has built-in support for displaying these errors. You see a red outline around controls that bind to the property in error.  UWP and Xamarin do not have built-in visuals (yet) and you will have to create them yourself. 

```
    public interface INotifyDataErrorInfo
    {
        // Gets a value that indicates whether the entity has validation errors. 
        bool HasErrors { get; }

        // Occurs when the validation errors have changed for a property or for the entire entity. 
        event EventHandler<DataErrorsChangedEventArgs> ErrorsChanged;

        // Gets the validation errors for a specified property or for the entire entity.
        IEnumerable GetErrors(string propertyName);
    }
```