convertersystems/opc-ua-samples #31: Server certificate rejected

andrea5586, 2017-05-09:
I recently heard about this project and I donwloaded it this morning. I compiled and run Workstation.RobotServer and I'm trying to connect to it using Workstation.ConsoleApp.
Unfortunatelly I get: 

```
Step 1 - Describe this app.
Step 2 - Create a certificate store.
Step 3 - Create a session with your server.
Error connecting to endpoint. Did the server reject our certificate?
Could not verify security on OpenSecureChannel request.
Press any key to close the program...
```

The trace from visual studio output is:
```
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: Discovering endpoints of 'opc.tcp://portatilesys8:26543/Workstation.RobotServer'.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Channel opening.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending OpenSecureChannelRequest, Handle: 1
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received OpenSecureChannelResponse, Handle: 1 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Channel opened.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending GetEndpointsRequest, Handle: 2
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Installed new security token 1.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received GetEndpointsResponse, Handle: 2 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Channel closing.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending CloseSecureChannelRequest, Handle: 3
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Channel closed.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Success discovering endpoints of 'opc.tcp://portatilesys8:26543/Workstation.RobotServer'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel opening.
'Workstation.ConsoleApp.exe' (CLR v4.0.30319: Workstation.ConsoleApp.exe): caricamento di 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.resources\v4.0_4.0.0.0_it_b77a5c561934e089\System.resources.dll' completato. Modulo compilato senza simboli.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Found certificate with subject alt name 'urn:PORTATILESYS8:MyHomework'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: Opening session channel with endpoint 'opc.tcp://portatilesys8:26543/Workstation.RobotServer'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityPolicy: 'http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityMode: 'SignAndEncrypt'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: UserIdentity: 'Workstation.ServiceModel.Ua.AnonymousIdentity'.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending OpenSecureChannelRequest, Handle: 1
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Error: Error receiving response. Could not verify security on OpenSecureChannel request.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel faulted.
Il thread 0x3e94 è terminato con il codice 0 (0x0).
Il thread 0x32bc è terminato con il codice 0 (0x0).
Eccezione generata: 'Workstation.ServiceModel.Ua.ServiceResultException' in mscorlib.dll
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel closing.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel closed.
Eccezione generata: 'Workstation.ServiceModel.Ua.ServiceResultException' in mscorlib.dll
Eccezione generata: 'Workstation.ServiceModel.Ua.ServiceResultException' in mscorlib.dll
```

If I connect to Workstation.RobotServer with UaExpert client and it works properly and I'm able to see the robot axis.

Am I missing some configuration?

awcullen, 2017-05-09:
Thank you for reporting this issue!
The application certificate was automatically created on first run, and was set with the wrong date. I should have used UTC time.
I have uploaded package 2.0.0-RC6 that corrects this problem.

andrea5586, 2017-05-10:
Great! Now it works! Congratulation for all your repos! I consider them extremely useful!

closed 2017-05-10 by andrea5586