convertersystems/opc-ua-samples #32: How to make these parameter can modify

Savoyyang, 2017-05-11:
Dear sir, 
       How to make these parameter can modify in View, seemingly cannot use class variables.

[Subscription(endpointUrl: "opc.tcp://192.168.11.18:4840", publishingInterval: 500, keepAliveCount: 20)]

awcullen, 2017-05-11:
You can change the endpointUrl using the 'map' method of the UaApplicationBuilder.
I'll give two examples. First, you can add a .Map(x, y) method to the builder that will replace all requests for endpoint 'x' with endpoint 'y' instead.   
```
            // Build and run an OPC UA application instance.
            this.application = new UaApplicationBuilder()
                .UseApplicationUri($"urn:{Dns.GetHostName()}:Workstation.MobileHmi")
                .UseDirectoryStore(Path.Combine(
                    Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData),
                    "pki"))
                .UseIdentity(this.ShowSignInDialog)
                .UseLoggerFactory(this.loggerFactory)
                .Map("opc.tcp://localhost:26543", "opc.tcp://10.0.2.2:26543")
                .Build();

            this.application.Run();
```

Second, you can add a .Map(configuration) method to the builder that will read a text file for the mappings.   

```
            // Read 'appSettings.json' for endpoint configuration
            var config = new ConfigurationBuilder()
                .SetBasePath(Directory.GetCurrentDirectory())
                .AddJsonFile("appSettings.json", true)
                .Build();

            // Build and run an OPC UA application instance.
            this.application = new UaApplicationBuilder()
                .UseApplicationUri($"urn:{Dns.GetHostName()}:Workstation.RobotHmi")
                .UseDirectoryStore(Path.Combine(
                    Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData),
                    "Workstation.RobotHmi",
                    "pki"))
                .UseIdentity(this.ShowSignInDialog)
                .UseLoggerFactory(this.loggerFactory)
                .Map(config)
                .Build();

            this.application.Run();
```

Lastly, with either mapping can specify the SecurityPolicyUri, which will select an endpoint with matching policy.

Savoyyang, 2017-05-11:
First, That appSettings.json can contain more than one endpointUrl ? and can contain  publishingInterval keepAliveCount ? If can,you can give me a examples of appSetting.json.

Second, Would use the first .Map("opc.tcp://localhost:26543", "opc.tcp://10.0.2.2:26543")  rather than [Subscription(endpointUrl: "opc.tcp://192.168.11.18:4840"]，
Or What is the relationship between them?

awcullen, 2017-05-11:
appSettings.json
```
{
  "maps": [
    {
      "requestedUrl": "opc.tcp://localhost:26543",
      "endpoint": {
        "endpointUrl": "opc.tcp://andrew-think:26543",
        "securityPolicyUri": "http://opcfoundation.org/UA/SecurityPolicy#None"
      }
    }
  ]
}
```

publishingInterval and keepAliveCount are not configurable.

Savoyyang, 2017-05-12:
I know this code is in the inside of the RobotHmi ,I forgot I was going to use is on the android App, Don't know whether you have the plan add this in Android App(and the SignInDialog next week?).
Thank you very much.

closed 2017-06-05 by awcullen