convertersystems/opc-ua-samples #33: [RobotApp] XAML Designer crashing caused by AlarmConditionToStringConverter  (VS2017)

rudoc, 2017-05-11:
Dear Andrew!
I know, you run project on VS2015, but maybe you will be interested how it runs on VS2017 as well.
This is about RobotApp.
I able to compile the project and run it, but facing with the XAML Designer crashing (Designer crash occurs only for MainPage with the following error:
Microsoft.MetadataReader.UnresolvedAssemblyException
Type universe cannot resolve assembly: Workstation.UaClient, Version=2.0.0.0, Culture=neutral, PublicKeyToken=null.). 
I noticed this when started to use RC version of UaClient, and until 1.5.12 there was no any issues.  
I found, that excluding reference of AlarmConditionToStringConverter from  <Page.Resources> solving the issue. Of cause, I had to comment <ItemsControl>…</ItemsControl> on page in order it to run.
It might be VS2017 setup environment though.

awcullen, 2017-05-11:
Today I am using VS2017 as well and having good experience.
I haven't seen that issue with the designer. I have two suggestions. 
First, the designer has a tiny button near the zoom percentage that allows you to "Disable Project Code".

Second, you may restart the Designer by selecting 'XDesProc.exe' in Task Manager's Details tab and clicking 'End Task'.

rudoc, 2017-05-12:
Thanks Andrew!
Both options are not helped for preventing XAML Designer crushing.
Similar problem was observed for number of users with no obvious reason on VS2015.
I will go for clean install, what was one of the solutions and will report the result.

rudoc, 2017-05-12:
Clean installation of VS2017 did not helped. I've got the same errors. I am puzzled. even though I could build and run the project, the XAML Designer crushed. 
[image]

If I would disable running project code I will get the UnresolvedAssemblyException error.

Dear Andrew! As it seems to be setup problem, please close the issue at your convenience.

awcullen, 2017-05-12:
Now I am getting the same errors as you have reported. Lets keep the issue open.

rudoc, 2018-08-12:
Latest VS2017 has no issues with design time representation

closed 2018-08-12 by rudoc