convertersystems/opc-ua-samples #34: OPC UA Example of website

tejasklodha, 2017-05-12:
Dear Sir,

**It is not issue but only suggestions...**

`I am Very Happy to use and learn about OPC UA on your repository ... `

All type of example available in your repository **except Website.** 
I think you also give some basic example that how we can see live data efficiently of OPC UA on Web...

Are you plan something like that?

Thanks,

awcullen, 2017-05-12:
I have not considered a cloud based service or a webclient. 
There are two projects I am watching in this category.

https://mva.microsoft.com/en-US/training-courses/introduction-to-azure-iot-17611

https://thingsboard.io/

tejasklodha, 2017-05-12:
Dear Sir,

Thanks for provide a links....

But I already successfully send OPC UA server data to azure portal and also show live data on web portal which is host on azure....
But disadvantage of cloud is that we pay for it even if We want to monitor data on **same network**...

**And beauty of you all applications is that we can able to use without cost If it is in same network....** So I Found Example of webapplication which is same network(Web site host on internal server)

Thanks,

abrasat, 2017-05-19:
Does the thingsboard platform allow also to write values to opc-ua nodes, or is just read access ?

awcullen, 2017-05-19:
Today the OpcUa nodes are readonly. They say writing to nodes is planned for next version.

abrasat, 2017-05-19:
Ok, thanks. Could you get it working with the RobotServer from the opc-ua samples ?

Savoyyang, 2017-05-22:
I also want to know @awcullen

closed 2019-01-25 by tejasklodha