convertersystems/opc-ua-samples #35: OPC UA samples for opc-ua client 2.0.0* prerelease

abrasat, 2017-05-16:
Are there any opc-ua samples compatible with the new opc-ua client 2.0.0* prerelease available ?
When do you plan to "officialize" the opc-ua client 2.0.0 version ?

awcullen, 2017-05-16:
The samples in this repo are using the latest 2.0.0* pre-release versions.

Do you like the new api? I am looking for feedback before "officializing".

tejasklodha, 2017-05-16:
@awcullen ... New version is very good and also easy to understand....

I m using it without any issue....

Tejas Lodha

abrasat, 2017-05-16:
Does the new version support also ExpandedNodeIds as MonitoredItemAttributes ?

abrasat, 2017-05-17:
I have an application where I used the UaTcpSessionClient and the Subscribe() method to register the monitored items from a ViewModel. What is the equivalent code with the new opc-ua client version, as UaTcpSessionClient does not exist anymore?
Also for the UaTcpSessionClient I selected "manually" which endpoint to connect (I did not want to always have the "default" one with the highest security)

        // discover available endpoints of server.
         var getEndpointsRequest = new GetEndpointsRequest
          {
                EndpointUrl = this.discoveryUrl,
                 ProfileUris = new[] { TransportProfileUris.UaTcpTransport }
          };
          this.logger.LogDebug($"Discovering endpoints of '{getEndpointsRequest.EndpointUrl}'.");
          var getEndpointsResponse = UaTcpDiscoveryClient.GetEndpointsAsync(getEndpointsRequest).Result;
          var selectedEndpoint = getEndpointsResponse.Endpoints.OrderByDescending(e => e.SecurityLevel).Last();

          this.session = new UaTcpSessionClient(
              new ApplicationDescription()
              {
	            ApplicationName = "AlarmUI",
	            ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:OpcUa.AlarmUI",
	            ApplicationType = ApplicationType.Client
              },
              new DirectoryStore(Path.Combine(Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData), "pki"),
                  loggerFactory: this.loggerFactory),
              this.ProvideUserIdentity,
              selectedEndpoint,
              this.loggerFactory);

How can the endpoint be selected with the new opc-ua client version, without using UaTcpSessionClient ?

abrasat, 2017-05-18:
What exactly is the meaning of/difference between the requestedUrl and the endpointUrl parameters from the Map() method in UaApplicationBuilder ?

awcullen, 2017-05-19:
In the MobileHmi sample, I added:
```
.Map("opc.tcp://localhost:26543", "opc.tcp://10.0.2.2:26543")
```
That was because I wanted requests for "opc.tcp://localhost:26543" to be directed to "opc.tcp://10.0.2.2:26543". 
(There is also a version that takes a config file, see sample RobotHmi).
If you don't set the SecurityPolicyUri, then the endpoint will be selected with the highest security.
If you also set SecurityPolicyUri (for example to "http://opcfoundation.org/UA/SecurityPolicy#None"), then you will find that the endpoint selected will be restricted to only endpoints with "None" security.
In you case maybe the 'None' security was turned off at the server?

abrasat, 2017-05-19:
And what parameters should be put for instance in the Map() call, if I want to access a server with the address "opc.tcp://192.168.1.1:4444", and choose the  SecurityPolicyUri "http://opcfoundation.org/UA/SecurityPolicy#None", but without any redirection ?
Where/how can be set the values for SecurityLevel, SecurityMode, SecurityPolicyUri ?

awcullen, 2017-05-19:
You could use:
```
                .Map("opc.tcp://192.168.1.1:4444", "opc.tcp://192.168.1.1:4444", SecurityPolicyUris.None)
```

closed 2017-06-05 by awcullen