convertersystems/opc-ua-samples #36: Get node id, name (tag name) in data change notification

balamcsd, 2017-06-27:
I am referring to the code in the below link :
**https://github.com/convertersystems/opc-ua-samples/blob/master/ConsoleApp/Program.cs**
When monitoring for multiple nodes (Tags), how to get the node name, node id etc in data changed notification.
Going to pass the node name (Tag name), its value and the timestamp to a rest API.
I have to get it in below lines of code :
foreach (var dcn in dcns)
{
foreach (var min in dcn.MonitoredItems)
{
Console.WriteLine($"clientHandle: {min.ClientHandle}; value: {min.Value}");
}
}

awcullen, 2017-06-27:
Hello,

When you prepare the MonitoredItemCreateRequest, choose a ClientHandle (UInt) that identifies this MonitoredItem later in the PublishResponse

You could keep a dictionary<UInt, NodeId> that allows you to look up the NodeId from the ClientHandle

```
                var itemsToCreate = new MonitoredItemCreateRequest[]
                {
                    new MonitoredItemCreateRequest
                    {
                        ItemToMonitor = new ReadValueId
                        {
                            NodeId = NodeId.Parse("i=2258"),
                            AttributeId = AttributeIds.Value
                        },
                        MonitoringMode = MonitoringMode.Reporting,
                        RequestedParameters = new MonitoringParameters
                        {
                            ClientHandle = 12345,  // Choose a number that identifies this MonitoredItem 
                                                                  // in the PublishResponse.
                            SamplingInterval = -1,
                            QueueSize = 0,
                            DiscardOldest = true
                        }
                    }
                };
```

balamcsd, 2017-06-28:
Sir,
Thanks for your quick response.
Does it mean that for each node (eg. Tag) being monitored we need to give different ClientHandle
and based on the ClientHandle we need to identify the MonitoredItem (Tag)
What if the nodes (Tags) being monitored are dynamic?
Is this only solution available now?
Thanks & regards

awcullen, 2017-06-28:
Yes, each MonitoredItem for a Subscription needs a unique ClientHandle.
The only data returned in 'PublishResponse' is SubscriptionId, ClientHandle, and DataValue (or EventList)

balamcsd, 2017-06-29:
I have followed what your suggested approach (dictionary) and succeeded.
One more query :
In Step 7 - Subscribe to PublishResponse stream. 
Want to move the data change notification part alone to a separate event handler.
Can you please suggest how to do that.
Thanks & regards