convertersystems/opc-ua-samples #37: OPC-UA Get the nodeid corresponding to Tag name and vice versa

balamcsd, 2017-06-28:
Sir,
I have Tagname 'Tag1' created in OPC server. (in real scenario someone will create)
For monitoring this tag for data change notification i need to specify nodeid in the format (i=2256)
Can you please post C# code to get the nodeid from tagname and vice versa
Thanks in advance for your help
Regards

new MonitoredItemCreateRequest { ItemToMonitor = new ReadValueId { NodeId = NodeId.Parse("i=2256"), AttributeId = AttributeIds.Value }, MonitoringMode = MonitoringMode.Reporting, RequestedParameters = new MonitoringParameters { ClientHandle = 12345, SamplingInterval = -1, QueueSize = 0, DiscardOldest = true } }

awcullen, 2017-06-28:
You can use software tools to browse the OPC Server namespace for 'Tag1'. One popular tool is 'UaExpert' by Unified Automation.  With UaExpert you can find 'Tag1' (the DisplayName) and get additional information about the node, like NodeId and DataType.

balamcsd, 2017-06-29:
I saw some service 'TranslateBrowsePathsToNodeIds() ' and i am not clear how to use it
In my case input is Tagname. By using this service, i have to get the nodeid (it can be vice versa also)
Request to please post c# code for 'TranslateBrowsePathsToNodeIds() '

awcullen, 2017-06-30:
I suggest you read the pdf [Part 4- Services](https://opcfoundation.org/developer-tools/specifications-unified-architecture/part-4-services/). You will find a lot of good information.

balamcsd, 2017-06-30:
Yes i will read that document.
Currently i am using Workstation.UaClient from nuget for creating a opc uaclient application in C#
Is this correct?

balamcsd, 2017-07-11:
I got a solution for this by recursive method.
Just posting the c# code so it will be useful for someone who is reading this post -

    public class Tag
    {
        public string TagName { get; set; }
        public string NodeId { get; set; }
    }
private Dictionary<uint, Tag> TagDatabase = new Dictionary<uint, Tag>();

                    Tag tag = new C2Tag();

                    tag .TagName = "Tag1";

                    TagDatabase.AddTag(tag );

                    Tag tag = new C2Tag();

                    tag .TagName = "Tag2";

                    TagDatabase.AddTag(tag );

        private uint GetClientHandleByTag(string TagName)
        {
            uint tagKey = 0;

            tagKey = TagDatabase.FirstOrDefault(x => x.Value.TagName == TagName).Key;

            return tagKey;
        }

 private static uint clientHandleId = 1;

        public void AddTag(C2Tag tag)
        {
                TagDatabase.Add(clientHandleId++, tag);
        }

        public void UpdateNodeByClientHandle(uint clientHandle, string nodeId)
        {
            TagDatabase[clientHandle].NodeId = nodeId;
        }

        private static async Task RecursivelyFindNode(UaTcpSessionChannel channel, NodeId nodeid)
        {
            BrowseRequest browseRequest = new BrowseRequest
            {
                NodesToBrowse = new BrowseDescription[] { new BrowseDescription { NodeId = nodeid, BrowseDirection = BrowseDirection.Forward, ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences), NodeClassMask = (uint)NodeClass.Variable | (uint)NodeClass.Object, IncludeSubtypes = true, ResultMask = (uint)BrowseResultMask.All } },
            };

            BrowseResponse browseResponse = await channel.BrowseAsync(browseRequest);

            foreach (var rd1 in browseResponse.Results[0].References ?? new ReferenceDescription[0])
            {

                uint chid = TagDatabase.GetClientHandleByTag(rd1.DisplayName.ToString());

                if (chid > 0)
                {
                    TagDatabase.UpdateNodeByClientHandle(chid, rd1.NodeId.ToString());
                }

                await RecursivelyFindNode(channel, ExpandedNodeId.ToNodeId(rd1.NodeId, channel.NamespaceUris));
            }
        }

To call recursive method -

await RecursivelyFindNode(channel, NodeId.Parse(ObjectIds.RootFolder));

balamcsd, 2017-07-11:
Based on above solution closing the issue

closed 2017-07-11 by balamcsd