convertersystems/opc-ua-samples #38: Can use thingsboard connect your UA server？

Savoyyang, 2017-06-28:
Sir,
I know thingsboard example can connect to kepware ua server, I try connect to server failed ,Have you tried to connect your server?

awcullen, 2017-07-08:
Yes I have.
The config file I used is:

[opc-config.json.txt](https://github.com/convertersystems/opc-ua-samples/files/1132623/opc-config.json.txt)

When the Robot1 device is connected, it will look like this:
[image]
You can make a dashboard like this:
[image]

Savoyyang, 2017-07-08:
I use losthost:8080 and The official account Tenant , so what's your tb-gateway.yml configuration？

awcullen, 2017-07-08:
Heres my yml file.
[tb-gateway.yml.txt](https://github.com/convertersystems/opc-ua-samples/files/1133096/tb-gateway.yml.txt)

  Please see [instructions](https://thingsboard.io/docs/iot-gateway/getting-started/#step-4-gateway-configuration) about setting the accessToken that is correct for your installation.

Savoyyang, 2017-07-08:
[opc-config.json.txt](https://github.com/convertersystems/opc-ua-samples/files/1133106/opc-config.json.txt)

[tb-gateway.yml.txt](https://github.com/convertersystems/opc-ua-samples/files/1133112/tb-gateway.yml.txt)

I follow this instructions but failed，After the configuration isWindows: 
net stop tb-gateway
net start tb-gateway

[image]

[image]

awcullen, 2017-07-08:
I don't see any errors.
Are all the services and RobotServer on one computer?
Could you share the file:
\tb-gateway\logs\tb-gateway.log

Savoyyang, 2017-07-08:
Of course, services and RobotServer on one computer

[tb-gateway.wrapper.log.txt](https://github.com/convertersystems/opc-ua-samples/files/1133155/tb-gateway.wrapper.log.txt)

awcullen, 2017-07-08:
Is there a tb-gateway.log ?

tejasklodha, 2017-07-08:
Dear sir,

For your information
In past, I also tried to connect with things board....

And its worked fine with Andrew's project.....

Thanks,

Savoyyang, 2017-07-08:
there is a file tb-gateway.log.
I am reboot,but still can't connect.
I think I should use other computer and try it again.
thanks.

awcullen, 2017-07-08:
Please share with us the tb-gateway.log

Savoyyang, 2017-07-08:
Under the path just have tb-gateway.wrapper.log.
[image]

awcullen, 2017-07-08:
Okay, then could you a cmd window (run as administrator).
Goto \tb-gateway
run: install.bat
run: net start tb-gateway

look for a \logs\tb-gateway.log

Savoyyang, 2017-07-08:
2017-07-09 02:05:35,908 [main] WARN  o.t.g.service.MqttGatewayService - Failed to connect to Thingsboard!
org.eclipse.paho.client.mqttv3.MqttException: 已断开连接
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:164)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.io.EOFException: null
	at java.io.DataInputStream.readByte(Unknown Source)
	at org.eclipse.paho.client.mqttv3.internal.wire.MqttInputStream.readMqttWireMessage(MqttInputStream.java:92)
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:116)
	... 1 common frames omitted
2017-07-09 02:05:40,110 [main] WARN  o.t.g.service.MqttGatewayService - Failed to connect to Thingsboard!
org.eclipse.paho.client.mqttv3.MqttException: 已断开连接
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:164)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.io.EOFException: null
	at java.io.DataInputStream.readByte(Unknown Source)
	at org.eclipse.paho.client.mqttv3.internal.wire.MqttInputStream.readMqttWireMessage(MqttInputStream.java:92)
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:116)
	... 1 common frames omitted

awcullen, 2017-07-08:
net start thingsboard

Savoyyang, 2017-07-08:
2017-07-09 02:10:18,973 [main] INFO  o.t.gateway.GatewayApplication - Starting GatewayApplication v1.2.0 on DESKTOP-1JD0SQ9 with PID 5412 (C:\Users\\Downloads\tb-gateway-windows-1.2\tb-gateway\lib\tb-gateway.jar started by DESKTOP-1JD0SQ9$ in C:\Users\\Downloads\tb-gateway-windows-1.2\tb-gateway\conf)
2017-07-09 02:10:18,979 [main] INFO  o.t.gateway.GatewayApplication - No active profile set, falling back to default profiles: default
2017-07-09 02:10:19,194 [main] INFO  o.s.b.c.e.AnnotationConfigEmbeddedWebApplicationContext - Refreshing org.springframework.boot.context.embedded.AnnotationConfigEmbeddedWebApplicationContext@6adca536: startup date [Sun Jul 09 02:10:19 CST 2017]; root of context hierarchy
2017-07-09 02:10:19,290 [background-preinit] INFO  o.h.validator.internal.util.Version - HV000001: Hibernate Validator 5.2.4.Final
2017-07-09 02:10:22,922 [main] INFO  o.s.b.c.e.t.TomcatEmbeddedServletContainer - Tomcat initialized with port(s): 9090 (http)
2017-07-09 02:10:22,949 [main] INFO  o.a.catalina.core.StandardService - Starting service Tomcat
2017-07-09 02:10:22,951 [main] INFO  o.a.catalina.core.StandardEngine - Starting Servlet Engine: Apache Tomcat/8.5.6
2017-07-09 02:10:23,655 [localhost-startStop-1] INFO  o.a.c.c.C.[Tomcat].[localhost].[/] - Initializing Spring embedded WebApplicationContext
2017-07-09 02:10:23,659 [localhost-startStop-1] INFO  o.s.web.context.ContextLoader - Root WebApplicationContext: initialization completed in 4469 ms
2017-07-09 02:10:23,900 [localhost-startStop-1] INFO  o.s.b.w.s.ServletRegistrationBean - Mapping servlet: 'dispatcherServlet' to [/]
2017-07-09 02:10:23,913 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'characterEncodingFilter' to: [/*]
2017-07-09 02:10:23,914 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'hiddenHttpMethodFilter' to: [/*]
2017-07-09 02:10:23,921 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'httpPutFormContentFilter' to: [/*]
2017-07-09 02:10:23,921 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'requestContextFilter' to: [/*]
2017-07-09 02:10:24,318 [main] INFO  o.t.g.s.c.TbPersistenceConfiguration - Initializing file persistence using directory: storage
2017-07-09 02:10:25,916 [main] WARN  o.t.g.service.MqttGatewayService - Failed to connect to Thingsboard!
org.eclipse.paho.client.mqttv3.MqttException: 已断开连接
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:164)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.io.EOFException: null
	at java.io.DataInputStream.readByte(Unknown Source)
	at org.eclipse.paho.client.mqttv3.internal.wire.MqttInputStream.readMqttWireMessage(MqttInputStream.java:92)
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:116)
	... 1 common frames omitted
2017-07-09 02:10:26,603 [pool-1-thread-1] INFO  o.t.g.s.update.DefaultUpdateService - New version 1.2.1 is available!

2017-07-09 02:12:20,135 [main] WARN  o.t.g.service.MqttGatewayService - Failed to connect to Thingsboard!
org.eclipse.paho.client.mqttv3.MqttException: 已断开连接
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:164)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.io.EOFException: null
	at java.io.DataInputStream.readByte(Unknown Source)
	at org.eclipse.paho.client.mqttv3.internal.wire.MqttInputStream.readMqttWireMessage(MqttInputStream.java:92)
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:116)
	... 1 common frames omitted
2017-07-09 02:12:24,158 [main] WARN  o.t.g.service.MqttGatewayService - Failed to connect to Thingsboard!
org.eclipse.paho.client.mqttv3.MqttException: 已断开连接
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:164)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.io.EOFException: null
	at java.io.DataInputStream.readByte(Unknown Source)
	at org.eclipse.paho.client.mqttv3.internal.wire.MqttInputStream.readMqttWireMessage(MqttInputStream.java:92)
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:116)
	... 1 common frames omitted

awcullen, 2017-07-08:
Wow. Lets go back to the instructions [here](https://thingsboard.io/docs/iot-gateway/getting-started/#step-3-gateway-provisioning).

Make a new device named 'Gateway'
Check the IsGateway box.
Copy the AccessToken.

open \tb-gateway\conf\tb-gateway.yml
paste the accessToken.
save

net stop tb-gateway
net start tb-gateway

Savoyyang, 2017-07-08:
[image]

Savoyyang, 2017-07-08:
feel so bad ，still failed to connect to Thingsboard.

awcullen, 2017-07-08:
Please share new tb-gateway.yml 
and tb-gateway.log

Savoyyang, 2017-07-08:
2017-07-09 02:58:38,387 [main] INFO  o.t.gateway.GatewayApplication - Starting GatewayApplication v1.2.0 on DESKTOP-1JD0SQ9 with PID 4244 (C:\Users\Ү˳أ\Downloads\tb-gateway-windows-1.2\tb-gateway\lib\tb-gateway.jar started by DESKTOP-1JD0SQ9$ in C:\Users\Ү˳أ\Downloads\tb-gateway-windows-1.2\tb-gateway\conf)
2017-07-09 02:58:38,398 [main] INFO  o.t.gateway.GatewayApplication - No active profile set, falling back to default profiles: default
2017-07-09 02:58:38,659 [main] INFO  o.s.b.c.e.AnnotationConfigEmbeddedWebApplicationContext - Refreshing org.springframework.boot.context.embedded.AnnotationConfigEmbeddedWebApplicationContext@6adca536: startup date [Sun Jul 09 02:58:38 CST 2017]; root of context hierarchy
2017-07-09 02:58:38,817 [background-preinit] INFO  o.h.validator.internal.util.Version - HV000001: Hibernate Validator 5.2.4.Final
2017-07-09 02:58:42,227 [main] INFO  o.s.b.c.e.t.TomcatEmbeddedServletContainer - Tomcat initialized with port(s): 9090 (http)
2017-07-09 02:58:42,253 [main] INFO  o.a.catalina.core.StandardService - Starting service Tomcat
2017-07-09 02:58:42,262 [main] INFO  o.a.catalina.core.StandardEngine - Starting Servlet Engine: Apache Tomcat/8.5.6
2017-07-09 02:58:42,685 [localhost-startStop-1] INFO  o.a.c.c.C.[Tomcat].[localhost].[/] - Initializing Spring embedded WebApplicationContext
2017-07-09 02:58:42,690 [localhost-startStop-1] INFO  o.s.web.context.ContextLoader - Root WebApplicationContext: initialization completed in 4031 ms
2017-07-09 02:58:43,053 [localhost-startStop-1] INFO  o.s.b.w.s.ServletRegistrationBean - Mapping servlet: 'dispatcherServlet' to [/]
2017-07-09 02:58:43,122 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'characterEncodingFilter' to: [/*]
2017-07-09 02:58:43,123 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'hiddenHttpMethodFilter' to: [/*]
2017-07-09 02:58:43,130 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'httpPutFormContentFilter' to: [/*]
2017-07-09 02:58:43,130 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'requestContextFilter' to: [/*]
2017-07-09 02:58:43,551 [main] INFO  o.t.g.s.c.TbPersistenceConfiguration - Initializing file persistence using directory: storage
2017-07-09 02:58:44,963 [main] WARN  o.t.g.service.MqttGatewayService - Failed to connect to Thingsboard!
org.eclipse.paho.client.mqttv3.MqttException: ӑ׏ߪlޓ
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:164)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.io.EOFException: null
	at java.io.DataInputStream.readByte(Unknown Source)
	at org.eclipse.paho.client.mqttv3.internal.wire.MqttInputStream.readMqttWireMessage(MqttInputStream.java:92)
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:116)
	... 1 common frames omitted

Savoyyang, 2017-07-08:
[tb-gateway.yml.txt](https://github.com/convertersystems/opc-ua-samples/files/1133230/tb-gateway.yml.txt)

awcullen, 2017-07-08:
I don't know the differences, but there is a newer release of tb-gateway. 
v 1.2.1 is available here: [releases](https://github.com/thingsboard/thingsboard-gateway/releases)

Savoyyang, 2017-07-08:
I found my mistake, the new version and old version conflict, so I uninstall the old version, install the new version,it success.I need to be more careful!
thank you very much!

awcullen, 2017-07-08:
Now you are an Expert!

closed 2017-07-08 by awcullen