convertersystems/opc-ua-samples #41: Unable to connect to RobotServer using ConsoleApp due to Certificate issue

DevProLog, 2017-10-24:
Hi,

I just forked the samples repo and simply tried to run the RobotServer and the ConsoleApp. In step 3 of the ConsoleApp, the app gets stuck. After about 15 seconds I get this in the console:
```
Step 1 - Describe this app.
Step 2 - Create a certificate store.
Step 3 - Create a session with your server.
Error connecting to endpoint. Did the server reject our certificate?
Could not verify security on OpenSecureChannel request.
Press any key to close the program...
```
Do I have to configure anything related to the certificates? Maybe in the '\opc-ua-samples\RobotServer\Workstation.RobotServer.Config.xml' config file?

I checked that the ConsoleApp created the following files:
`\AppData\Local\Workstation.ConsoleApp\pki\own\certs\********.crt`
`\AppData\Local\Workstation.ConsoleApp\pki\own\private\********.key`

Thanks!

awcullen, 2017-10-24:
I'm glad to see the ConsoleApp created it's client certificate. Next, let's check if the RobotServer accepted it. The RobotServer stores certificates it rejects in the directory:
C:\ProgramData\OPC Foundation\CertificateStores\RejectedCertificates\certs

If the ConsoleApp certificate is there, you can move it to:
C:\ProgramData\OPC Foundation\CertificateStores\UA Applications\certs

awcullen, 2017-10-24:
Also, please send the text from the Debug Output window. There may be more info there that can help. 
Andrew

DevProLog, 2017-10-25:
The following files are found in the server certificate folders:
```
\OPC Foundation\CertificateStores\UA Applications\certs\Connectivity Explorer [*******].der
\OPC Foundation\CertificateStores\UA Applications\certs\Workstation.RobotServer [********].der
\OPC Foundation\CertificateStores\MachineDefault\certs\Connectivity Explorer [*******].der
\OPC Foundation\CertificateStores\MachineDefault\certs\Workstation.RobotServer [********].der
\OPC Foundation\CertificateStores\MachineDefault\private\Connectivity Explorer [*******].pfx
\OPC Foundation\CertificateStores\MachineDefault\private\Workstation.RobotServer [********].pfx
```

Here's the debug output of the RobotServer:
```
'Workstation.RobotServer.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_64\mscorlib\v4.0_4.0.0.0__b77a5c561934e089\mscorlib.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'C:\Projects\Ecophos\WBSServices\DevTestProjects\DCS\MIT_Repos\opc-ua-samples\RobotServer\bin\Debug\Workstation.RobotServer.exe'. Symbols loaded.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\Projects\Ecophos\WBSServices\DevTestProjects\DCS\MIT_Repos\opc-ua-samples\RobotServer\bin\Debug\Opc.Ua.Configuration.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\Projects\Ecophos\WBSServices\DevTestProjects\DCS\MIT_Repos\opc-ua-samples\RobotServer\bin\Debug\Opc.Ua.Core.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Windows.Forms\v4.0_4.0.0.0__b77a5c561934e089\System.Windows.Forms.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System\v4.0_4.0.0.0__b77a5c561934e089\System.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Drawing\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Drawing.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\Projects\Ecophos\WBSServices\DevTestProjects\DCS\MIT_Repos\opc-ua-samples\RobotServer\bin\Debug\Opc.Ua.Server.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Runtime.Serialization\v4.0_4.0.0.0__b77a5c561934e089\System.Runtime.Serialization.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\SMDiagnostics\v4.0_4.0.0.0__b77a5c561934e089\SMDiagnostics.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Xml\v4.0_4.0.0.0__b77a5c561934e089\System.Xml.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.ServiceModel.Internals\v4.0_4.0.0.0__31bf3856ad364e35\System.ServiceModel.Internals.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Configuration\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Configuration.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Core\v4.0_4.0.0.0__b77a5c561934e089\System.Core.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.ServiceModel\v4.0_4.0.0.0__b77a5c561934e089\System.ServiceModel.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Workstation.RobotServer.exe' (CLR v4.0.30319: Workstation.RobotServer.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.IdentityModel\v4.0_4.0.0.0__b77a5c561934e089\System.IdentityModel.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
The thread 0x40d8 has exited with code 0 (0x0).
The thread 0x3fa8 has exited with code 0 (0x0).
The thread 0x48f4 has exited with code 0 (0x0).
[... cut ...]
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending PublishRequest, Handle: 7
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending PublishRequest, Handle: 8
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending PublishRequest, Handle: 9
The thread 0x5e98 has exited with code 0 (0x0).
The thread 0x4790 has exited with code 0 (0x0).
The thread 0x59e0 has exited with code 0 (0x0).
The thread 0x63d0 has exited with code 0 (0x0).
The thread 0x237c has exited with code 0 (0x0).
```

awcullen, 2017-10-25:
The debug output for the ConsoleApp shows the client opening a secure channel, creating a session, reading the namespace table, and creating a subscription (for keep-alive purposes).  If the client certificate was not accepted, then there would errors.

closed 2018-03-29 by awcullen