convertersystems/opc-ua-samples #44: Why does this throw "LocalPrivateKey is null"

basprins, 2019-05-08:
Hi Andrew,

Why is this throwing an exception that localprivatekey is null? I am trying to connect without any security.

        public OpcUaClient()
        {
            _appDescription = new ApplicationDescription()
            {
                ApplicationName = "Test",
                ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:Bla",
                ApplicationType = ApplicationType.Client,
            };
        }

        public async Task Connect()
        {
            const string discoveryUrl = "opc.tcp://localhost:4840"; 

            var appDescription = new ApplicationDescription()
            {
                ApplicationName = "Test",
                ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:Test",
                ApplicationType = ApplicationType.Client,
            };

            _channel = new UaTcpSessionChannel(
                appDescription,
                null, 
                async args => await Task.FromResult(new AnonymousIdentity()),
                discoveryUrl,
                loggerFactory:null);

            try
            {
                await _channel.OpenAsync();
            }
            catch (Exception e)
            {

            }

        }

awcullen, 2019-05-11:
Hi. When you open a SessionChannel with just the url, the channel first searches for the most secure endpoint that the server offers.  Add the 'securityPolicyUri' parameter to specify the channel should choose an endpoints that matches the policy 'None'  
```
_channel = new UaTcpSessionChannel(
            appDescription,
            null, 
            async args => await Task.FromResult(new AnonymousIdentity()),
            discoveryUrl,
            SecurityPolicyUris.None,   // <---
            loggerFactory:null);
```

basprins, 2019-05-13:
Ah djeeze at least im consitent in leaving out vital arguments.. Thx this is solved

closed 2019-05-13 by basprins